robotology/wearables #1: Software architecture

diegoferigo, 2018-06-05:
Issue #1! :tada: 

Let's use this issue to outline the class hierarchy for this project. You can find here below my first draft, feel free to edit it (we now have versioned commits!)

```cpp
namespace Wear { // Main namespace

     // Main interface    
    class IWear;

    // Implementations
    class XSensSuit; // : public IWear
    class EmgSuit;   // : public IWear
    class FTShoe;    // : public IWear

    namespace Sensor {

        // Main interface
        class ISensor;

        // Interfaces
        class IOrientation;   // 4D 
        class IPosition;      // 3D
        class IAccelerometer; // 3D
        class IGyroscope;     // 3D
        class IMagnetometer;  // 3D
        class ILinearVector;  // 3D + 3D + 3D
        class IAngularVector; // 4D (spherical) + 3D + 3D
        class ITorque;        // 6D
        class ISkin;
        class IEmg;

        // Derived interfaces
        class IVirtualLink;  // : public ILinearVector, IAngularVector
        class IVirtualJoint; // : public IAngularVector

        // Implementations
        class FreeBodyAcceleration; // : public IAccelerometer
        class FTShoe;               // : public ITorque

    } // Sensor
} // Wear
```

Then, what will match yarp interfaces in `yarp::dev`, the implementation will implement also that interface. An example for the orientation:

```cpp
class Wear::XSensSuit : public Wear::IWear,
                        public Wear::Sensor::IOrientation,
                        ...
                        public yarp::dev::IOrientationSensors
```

cc @lucaTagliapietra @claudia-lat

diegoferigo, 2018-06-06:
Ping @traversaro. I noticed you already passed here, but not the situation is more stable. Do you have any comments on our choices?

traversaro, 2018-06-06:
Ok in general. 

Some comments: 
 * `bool getTx(wear::DataType& tx);` not a big fan of `wear::DataType` in place of `double`, it quite  complicates the readability of the API 
* not a big fan of `Wrench` as a term, I think it is quite a robotics jargon that just complicates readability
and its precise meaning in screw theory does not match perfectly what we mean by force torque 
* I think `ITorque` can be confusing as people may associate Torque with "1D joint torques" instead of "3D torques".
* We need to agree on a precise definition for RPY, linear velocity, angular velocity, linear acceleration, angular acceleration, otherwise any integration will be a nightmare. 
* For input/output of vector passed by reference I think `std::span` as described in https://github.com/robotology/idyntree/issues/430 is the perfect solution, unfortunately it will not be available until C++20. Have you consider if it make sense to define `wear::Span` ?

diegoferigo, 2018-06-06:
> bool getTx(wear::DataType& tx); not a big fan of wear::DataType in place of double, it quite complicates the readability of the API . Even if you want to be general, I think it make sense to use a more readable name such as T or NumericType.

This is an attempt of having a centralized place which allows reducing the bandwidth if it is necessary. However, I'm not sure how to deal with it in thrift. If it is not possible, reverting to double would be the best choice.

> We need to agree on a precise definition for RPY, linear velocity, angular velocity, linear acceleration, angular acceleration, otherwise any integration will be a nightmare.

Yes, it should be stated clearly in the documentation.

> For input/output of vector passed by reference (and simplify actual use of the API) I think std::span as described in robotology/idyntree#430 is the perfect solution, unfortunately it will not be available until C++20. Have you consider if it make sense to define wear::Span ?

I'd really like to use span since there are **a lot** of data going around in this application. I'm not sure at this early stage how to deal and which problem we might face using external libraries in public interface headers. Any pointers or recommendations? 

Let's talk f2f on Torque / Wrench.

traversaro, 2018-06-06:
> I'd really like to use span since there are a lot of data going around in this application. I'm not sure at this early stage how to deal and which problem we might face using external libraries in public interface headers. Any pointers or recommendations?

I was just thinking of just copying some MIT-licensed implementation of the span class in wear, as you can see in https://github.com/robotology/idyntree/pull/434  in the actual logic in the span class is quite limited, and given that the class is already standardized I don't think there will be a lot of activity on that front.

diegoferigo, 2019-02-08:
The outline of the `IWear` interface and its sensors has been fixed and it's quite stable already since few months. You can find it in the following resources. Closing this issue.

- [IWear.h](https://github.com/robotology-playground/wearables/blob/master/interfaces/IWear/include/Wearable/IWear/IWear.h)
- [ISensor.h](https://github.com/robotology-playground/wearables/blob/master/interfaces/IWear/include/Wearable/IWear/Sensors/ISensor.h)
- [Sensors](https://github.com/robotology-playground/wearables/tree/master/interfaces/IWear/include/Wearable/IWear/Sensors)

closed 2019-02-08 by diegoferigo