robotology/wearables #11: Cleanup directory structure and namespaces

diegoferigo, 2018-07-03:
Proposal for the directory structure:

- **`drivers`** Contains all the drivers such as #6 #7
- **`interfaces`** Contains the `IWear` interface and the `IXsensMVNControl`
- **`wrappers`** Contains all the devices for exposing classes to the network, such as `IWearWrapper` #8, `IWearRemapper` #9, `WearDataToXsensFrameWrapper` #5
- **`msgs`** Contains all the shared thrift messages

For what concern namespaces, I would go for the following choice:

```cpp
namespace wearable {
    class IWear;

    namespace wrappers {
        class IWearWrapper;
        class IWearRemapper;
        class WearDataToXsensFrameWrapper;
    }

    namespace ??? {
        class XsensSuit;
        class FTShoes;
    }
}

namespace xsensmvn {
    class XsensMVNDriver;
    class XSensMVNCalibrator;
    class IXsensMVNControl;
}
```

lrapetti, 2019-02-06:
I think this activity has been completed within the [`feature/cleanup`](https://github.com/robotology-playground/wearables/tree/feature/cleanup).
Am I correct @diegoferigo? Can we close the issue?

lrapetti, 2019-02-06:
@diegoferigo 
Given that the `feature/cleanup` branch has been heavily tested, and new features are under development based on that branch, do you think we may do a PR to the`master`?

CC @traversaro

diegoferigo, 2019-02-07:
Yes we can close this issue. In my opinion before merging `feature/cleanup` to `master`, I would like to have the other edits almost ready (if well organized in different PRs with a clean git history would speed up the review process). Before opening the PRs we can merge `feature/cleanup` and base all of them on top of master.

closed 2019-02-07 by diegoferigo