robotology/wearables #14: Write README.md

diegoferigo, 2018-07-03:


lrapetti, 2018-11-20:
One thing that should be mentioned in the `README.md` is that it may be required to export the installation directory to `IWear_DIR` environment variable.

traversaro, 2018-11-20:
@lrapetti Alternativly you may just append the installation prefix to the `CMAKE_PREFIX_PATH` enviroment variable.

claudia-lat, 2020-11-09:
@lrapetti @Yeshasvitvs please understand if we have to close this issue.

lrapetti, 2020-11-16:
Preliminary documentation on how to use the wearable devices is there, we still miss documentation regarding the compilation. Probably we should add it before closing, I move this in backlog so that we can achieve it in the next Iteration.

lrapetti, 2020-11-24:
> Preliminary documentation on how to use the wearable devices is there, we still miss documentation regarding the compilation.

I have add this documentation with https://github.com/robotology/wearables/pull/96. I will close this issue, feel free to open new issues in case you think any essential information is still missing.

closed 2020-11-24 by lrapetti