robotology/wearables #15: WearData thrift message does not compile on Xenial

diegoferigo, 2018-07-03:
#### Describe the bug
For unknown reason there are issues in Ubuntu Xenial during the compilation of `SensorStatus.h`:

```
FAILED: /usr/lib/ccache/g++   -DIWearWrapper_EXPORTS -I../wrappers/IWear/include -I../IWear/.. -Imsgs/include -isystem /projects/robotology-superbuild/build/install/include -g -fPIC   -std=gnu++14 -MMD -MT wrappers/IWear/CMakeFiles/IWearWrapper.dir/src/IWearWrapper.cpp.o -MF wrappers/IWear/CMakeFiles/IWearWrapper.dir/src/IWearWrapper.cpp.o.d -o wrappers/IWear/CMakeFiles/IWearWrapper.dir/src/IWearWrapper.cpp.o -c ../wrappers/IWear/src/IWearWrapper.cpp
In file included from msgs/include/thrift/SensorInfo.h:17:0,
                 from msgs/include/thrift/Accelerometer.h:17,
                 from msgs/include/thrift/WearData.h:17,
                 from ../wrappers/IWear/src/IWearWrapper.cpp:11:
msgs/include/thrift/SensorStatus.h:24:6: error: expected identifier before numeric constant
       OVERFLOW = 3,
      ^
msgs/include/thrift/SensorStatus.h:24:6: error: expected '}' before numeric constant
msgs/include/thrift/SensorStatus.h:24:6: error: expected unqualified-id before numeric constant
In file included from msgs/include/thrift/SensorInfo.h:17:0,
                 from msgs/include/thrift/Accelerometer.h:17,
                 from msgs/include/thrift/WearData.h:17,
                 from ../wrappers/IWear/src/IWearWrapper.cpp:11:
msgs/include/thrift/SensorStatus.h:32:1: error: expected declaration before '}' token
 }
 ^
ninja: build stopped: subcommand failed.
```

#### Desktop (please complete the following information):
 - OS: Ubuntu 16.04
 - Compiler: GCC 5.4.0 / Clang 3.8.0

#### Additional information

On Ubuntu Bionic everything works fine.

diegoferigo, 2018-07-14:
Also on macOS @claudia-lat had the same issue. @lucaTagliapietra did you manage to solve it?

lrapetti, 2019-02-06:
Currently, @Yeshasvitvs, @kouroshD and I are all compiling `wearables` without receiving this error.
This error may be related to an older version of the software, or the fact that you were using older version of the compiler (`gcc-5` is currently not fully supported).

@diegoferigo Probably we may close this issue.

diegoferigo, 2019-02-07:
I remember it was failing on Xenial, and so far no one is using it as far as I know. We can close it, leaving still pending the definition of the minimum system requirements for running `wearables`.

closed 2019-02-07 by diegoferigo