robotology/wearables #3: Develop the new XsensMVNDriver class

diegoferigo, 2018-07-03:
This class will replace the former [robotology-playground/xsens-mvn/xsensdriver](https://github.com/robotology-playground/xsens-mvn/tree/master/xsensdriver).

Features:

- Support of the Xsens 2018.0.3 SDK
- YARP independent
- C++14
- Works only in Windows (due to Xsens' SDK)

lrapetti, 2019-02-06:
`XsensMVNDriver` has been developed (https://github.com/robotology-playground/wearables/tree/feature/cleanup/XSensMVN). 
@diegoferigo if you agree, we can close this issue.

traversaro, 2019-02-06:
I typically find more convenient to close issues when the relevant features has landed in `master` or `devel`, but I guess it is up to the maintainer to decide the policy on this.

lrapetti, 2019-02-06:
> I typically find more convenient to close issues when the relevant features has landed in `master` or `devel`, but I guess it is up to the maintainer to decide the policy on this.

I was pointing to a branch, but the same feature is also supported in the `master` (https://github.com/robotology-playground/wearables/tree/master/XSensMVN).

traversaro, 2019-02-06:
Great!

diegoferigo, 2019-02-07:
Yes, closing.

diegoferigo, 2019-02-07:
One missing thing is exposing the virtual points, as discussed with @claudia-lat. We will discuss it in a separate issue when we will need this feature and start the development.

closed 2019-02-07 by diegoferigo