robotology/wearables #7: Develop FTShoes

diegoferigo, 2018-07-03:
This new class is one of the implementations of the `IWear` interface. It will expose the data from the [ftShoe](https://github.com/robotology-playground/forcetorque-yarp-devices/tree/master/ftShoe), particularly:

- FT sensors
- IMUs
- Temperature sensors

lrapetti, 2019-02-06:
`FTShoes` has been developed (https://github.com/robotology-playground/wearables/tree/master/devices/FTShoes). 
@diegoferigo @Yeshasvitvs if you agree, we can close this issue.

diegoferigo, 2019-02-07:
In my opinion we should wait the development of the device that wraps `MultipleAnalogSensor` interfaces to an `IWear` interface. Right now only the FTSensors have been implemented, using the old `IAnalogSensor` interface.

lrapetti, 2020-01-27:
> In my opinion we should wait the development of the device that wraps `MultipleAnalogSensor` interfaces to an `IWear` interface. Right now only the FTSensors have been implemented, using the old `IAnalogSensor` interface.

This missing part is related to #18. Closing this issue since an implementation of FTShoe is available.

closed 2020-01-27 by lrapetti