robotology/wearables #9: Develop IWearRemapper

diegoferigo, 2018-07-03:
This class should provide the following features:

- Read multiple ports streaming `WearData`
- Combine the data of all the input ports
- Provide a single output containing the merged data
- Allow using it as a device by implementing the `IWear` interface

traversaro, 2019-01-28:
It may be worth to avoid the `IWearRemapper` name for this device, as w.r.t. to the `remapper` devices implemented in YARP (see http://www.yarp.it/group__dev__impl__remappers.html), this devices only supports remapping data that is read via the YARP network, and do not support "remapping" devices to which it is attached via `yarprobotinterface`. A possible name would be ` `IWearClientRemapper` or ` `IWearRemoteRemapper`, using classical names used for YARP network clients (see http://www.yarp.it/group__dev__impl__network__clients.html).

If we do not want to change its name for convenience, I suggest to explicitly document this in the `IWearRemapper` documentation.

lrapetti, 2019-02-06:
Since the implementation has been completed I think we can close this issue.
Concerning the name, we can document it and eventually change the name in a second instance (it should not require many changes since the usage of the devices is still limited to [`HDE`](https://github.com/robotology/human-dynamics-estimation)).

@diegoferigo @Yeshasvitvs

diegoferigo, 2019-02-07:
Yes, closing.

closed 2019-02-07 by diegoferigo