robotology/wearables #18: Implement MultipleAnalogSensor to IWear interface

lrapetti, 2019-02-07:
It has to be developed an interface that map a [`multipleAnalogSensor`](http://www.yarp.it/group__dev__iface__multiple__analog.html) interface, into `IWear` interface.

traversaro, 2019-02-07:
This would be extremly useful also to access all the distributed accelerometers and gyroscopes available in iCub, and make them available in wearables.  cc @nunoguedelha

Yeshasvitvs, 2019-02-15:
[image]

This is from the brief discussion @diegoferigo and I had about how to architect the FTShoes wearable device using [`multipleAnalogSensor`](http://www.yarp.it/group__dev__iface__multiple__analog.html) interface

lrapetti, 2019-02-15:
> This is from the brief discussion @diegoferigo and I had about how to architect the FTShoes wearable device using [`multipleAnalogSensor`](http://www.yarp.it/group__dev__iface__multiple__analog.html) interface

The insole skin is kept separate from the shoe interface, indeed not included here right?

diegoferigo, 2019-02-16:
@lrapetti The picture was sketched when the skin situation was not mature as it is today and it was a discussion that took into account also the amount of modifications.

In my opinion the best and more comlpete solution would be the one in the bottom left, but it would require:

- Update the device of the single FT sensor to expose MAS
- Update the device that merge the two FT sensors to read from MAS interfaces and expose a combined one
- Develop a device to map MAS to IWear

The first two should not be very demanding, the third one needs some work.

nunoguedelha, 2019-02-16:
@diegoferigo 
>     * Update the device of the single FT sensor to expose MAS

When Valentina Gaggero did this for the inertial sensors on iCub, she told me it was not a great effort. Maybe you can get some ideas looking at how it was done on that case.

>     * Update the device that merge the two FT sensors to read from MAS interfaces and expose a combined one
As far as I remember this can be done by a MAS-Remapper device and does not need additional work right? You just need to replace the old device by the remapper. I can help on the configuration since I did it for iCub.

diegoferigo, 2019-02-19:
> As far as I remember this can be done by a MAS-Remapper device and does not need additional work right?

I forgot its existence honestly. However, we cannot use it directly because we should embed also the logic to fuse the two readings from the front and rear FT sensors, and the access to the skin.

nunoguedelha, 2019-02-19:
@diegoferigo , @traversaro , this reminds me think of something that crossed my mind when I started working on the angular acceleration estimation. It would be interesting if we could attach to the remapper a device exposing a virtual sensor that actually fuses measurements from other real sensors that are also attached to the same remapper. This would be applicable for the use case @diegoferigo mentioned, and for instance the case for the angular acceleration estimation fusing accelerometers and gyroscopes measurements (this estimation algorithm is still under development).

(Note:

diegoferigo, 2019-02-19:
@nunoguedelha Programmatically adding new sensor types without editing the MAS interfaces is not possible in my opinion. All the methods inside the sensors interfaces have different name, in other words, there is no unified interface for all sensors. Hence, I do not think is possible using the remapper for a generic type.

Even though this issue is not the right place where to discuss this matter, the options are the following:

- If the sensor has a generic applicability, we should extend the MAS interfaces adding upstream the new sensor type
- Create ad-hoc interface and client / server architecture for the specific sensor

nunoguedelha, 2019-02-19:
ah yes, true, I had forgotten that there were a lot of sensor specific method names.

traversaro, 2019-02-19:
> Programmatically adding new sensor types without editing the MAS interfaces is not possible in my opinion. All the methods inside the sensors interfaces have different name, in other words, there is no unified interface for all sensors.

I am probably missing something, but why this is related to what @nunoguedelha suggested? The combined sensor (even if the "combination" logic can not be part of the generic `` device) will be again a FT sensor, that can be exposed using the existing MAS methods. 

> Hence, I do not think is possible using the remapper for a generic type.

I am not sure what you mean here by "generic type", but to avoid confusion for anyone reading: the [`multipleanalogsensorswrapper`](http://www.yarp.it/classyarp_1_1dev_1_1MultipleAnalogSensorsRemapper.html) is a remapper that supports **all** the sensors supported by MultipleAnalogSensors interfaces.

diegoferigo, 2019-02-19:
> The combined sensor (even if the "combination" logic can not be part of the generic `` device) will be again a FT sensor, that can be exposed using the existing MAS methods.

Yes I understood that he meant something related another application he had in mind. In this case there is a 2FT -> 1FT map so all's good.

> I am not sure what you mean here by "generic type"

Indeed, a type not supported by MAS

nunoguedelha, 2019-02-19:
> Yes I understood that he meant something related another application he had in mind. In this case there is a 2FT -> 1FT map so all's good.

my memory of the MAS interface was unclear. As @traversaro understood, my idea was to use several sensors for emulating a sensor type supported by the MAS interface. In my case that would be the type `IThreeAxisAngularAccelerometers`. If that type is not supported yet it should be added (that's the plan).