robotology/wearables #26: Update CI pipeline

diegoferigo, 2019-02-08:
Currently AppVeyor builds are broken. I just disabled them since their maintenance is quite demanding since we have to rely on the binary installers which sometimes they are not promptly updated.

I would recommend to switch to a windows stage in Travis. In this way single `install.sh` and `script.sh` have to be maintained, and will be used in both windows and macOS (actually, `script.sh` can be used also in linux stages).

Referring to [`.ci/install.sh`](https://github.com/robotology-playground/wearables/blob/master/.ci/install.sh), I don't remember honestly why we need `icub-main` and `icub-firmware-shared` dependencies. Maybe now just `yarp` would suffice. In the windows case, we should check what is the best way to install dependencies.

As reference, have a look at https://github.com/diegoferigo/blockfactory-demo-ros/blob/master/.travis.yml or https://github.com/robotology/blockfactory/blob/master/.travis.yml.

diegoferigo, 2019-02-08:
cc @lrapetti @Yeshasvitvs

lrapetti, 2020-01-28:
CI has been cleaned so I'll close this issue.
Missing CI for windows will be tracked by https://github.com/robotology/wearables/issues/77https://github.com/robotology/wearables/issues/77

closed 2020-01-28 by lrapetti