robotology/wearables #27: Data of a sensor is missing from a dumped IWearRemapper file for certain timestamps

claudia-lat, 2019-02-13:
By checking the data dumped from the IWearRemapper, I found the following behaviour: 

```bash
...
-1 27976.097323 IWearRemapper ... (("XsensSuit::fbAcc::Head" ("XsensSuit::fbAcc::Head" 1 -0.016373313533672 -0.0190550281415655 0.000531472733429794)) ("XsensSuit::fbAcc::LeftFoot" ("XsensSuit::fbAcc::LeftFoot" 1 0.00442949986499089 0.00220952626739518 0.01514176584128)) ...
-1 27976.105447 IWearRemapper ... (("XsensSuit::fbAcc::LeftFoot" ("XsensSuit::fbAcc::LeftFoot" 1 0.00722438204900699 0.00521013136724258 0.0151151371458731)) ...
-1 27976.110341 IWearRemapper ... (("XsensSuit::fbAcc::LeftFoot" ("XsensSuit::fbAcc::LeftFoot" 1 0.00722438204900699 0.00521013136724258 0.0151151371458731)) ...
-1 27976.119040 IWearRemapper ... (("XsensSuit::fbAcc::Head" ("XsensSuit::fbAcc::Head" 1 0.0190951423117848 -0.00909498756059363 -0.0739544957307423)) ("XsensSuit::fbAcc::LeftFoot" ("XsensSuit::fbAcc::LeftFoot" 1 -0.0328387372429226 -0.0157998343825085 -0.00982986474280878)) ...
-1 27976.131309 IWearRemapper ... (("XsensSuit::fbAcc::Head" ("XsensSuit::fbAcc::Head" 1 0.038067409596767 -0.00213547009733639 -0.018574056519352)) ("XsensSuit::fbAcc::LeftFoot" ("XsensSuit::fbAcc::LeftFoot" 1 -0.0300501728865379 -0.0128057789517119 -0.00983834574195441))...
...
```

The second and third samples don't have the `XsensSuit::fbAcc::Head`.

DanielePucci, 2019-02-18:
Please guys add the comments that we wrote on the chat. cc @Yeshasvitvs

Yeshasvitvs, 2019-02-19:
As @diegoferigo highlighted,

"However, if this happens only in the first few samples, a possible explanation is that after the calibration, some getSensorData (or similar method) is called in the xsens driver, but the driver still has to properly initialize itself. We might be too quick to ask data right after the calibration is finished"

In addition, we should assert that the wearable data size is equal to the  number of sensors.

claudia-lat, 2019-02-24:
In another dumped file I found another missing data, this time for the shoes:

```bash
...
-1 0.000000 1312.897323 IWearRemapper () () () ((FTShoeLeftFTSensors (FTShoeLeftFTSensors 1 1.20477376898739 17.3058125032916 365.895029706986 -3.00952605367758 -25.6593735630912 5.56826094317577)) (FTShoeRightFTSensors (FTShoeRightFTSensors 1 -0.176451816950106 19.1924397005572 331.252329511674 -2.8651473915682 -23.6995854163627 5.19620440316723))) ...
-1 0.000000 1312.900958 IWearRemapper () () () ((FTShoeLeftFTSensors (FTShoeLeftFTSensors 1 0.810120448674894 17.6638752474322 365.894205732377 -3.0077255165682 -25.7377736510734 5.60779743868969)))...
-1 0.000000 1312.918355 IWearRemapper () () () ((FTShoeLeftFTSensors (FTShoeLeftFTSensors 1 1.42492757758114 17.561519290401 365.243479414018 -3.01276091695883 -25.6533310826225 5.58480510610912)))...
-1 0.000000 1312.922112 IWearRemapper () () () ((FTShoeLeftFTSensors (FTShoeLeftFTSensors 1 1.42492757758114 17.561519290401 365.243479414018 -3.01276091695883 -25.6533310826225 5.58480510610912)) (FTShoeRightFTSensors (FTShoeRightFTSensors 1 -0.0464469341376059 19.601314212276 330.209849042924 -2.86670378805258 -23.6856083655815 5.20430415762279))) ...
...
```
Sometimes, the values of one shoes are not streamed at all.
cc @lrapetti @diegoferigo @Yeshasvitvs

DanielePucci, 2019-02-25:
First check is to see if the number of sensors and the wearable data size are similar; the second check is to investigate the synchronisation of all the sensors.

diegoferigo, 2019-02-25:
I already discussed this with someone, not sure whom. Currently the `IWearRemapper` device performs two operations. It reads `WearableData` messages from multiple sources and:

1. Forwards the message to an output port
1. Exposes and `IWear` interface from which sensors can be gathered

The first point is implemented as follows:

https://github.com/robotology-playground/wearables/blob/f6999a0f5f36161b4aa4bbdc32ec7bbe2f508092/devices/IWearRemapper/src/IWearRemapper.cpp#L353-L356

Note that this callback is executed every time a single-source message is read. This means that only the data coming from that source is overridden, and the rest instead stays untouched. Actually, since the callbacks can be executed in parallel if two messages from two sources arrive in the same moment, the output message can be mixed.

I just realized that the 2. is thread safe (there are mutexes in the default sensors implementation),  but not 1. If we want to make it thread safe, we should protect with a mutex also this section:

https://github.com/robotology-playground/wearables/blob/f6999a0f5f36161b4aa4bbdc32ec7bbe2f508092/devices/IWearRemapper/src/IWearRemapper.cpp#L364-L365

---

However, my original point is that the data forwarding to the output port is not an optimal solution. It was developed in the early stages of this project, and now there is a better solution. In my opinion the remapper should only expose an `IWear` interface. The logic about the message forwarding should be removed from the `IWearRemapper`, and instead it should be substituted by attaching to it a `IWearWrapper` device (which rate can be also changed).

lrapetti, 2019-02-25:
> In my opinion the remapper should only expose an `IWear` interface. The logic about the message forwarding should be removed from the `IWearRemapper`, and instead it should be substituted by attaching to it a `IWearWrapper` device (which rate can be also changed).

We tried to dump data through the `IWearRemapper` and we didn't experience any missing data (further tests in the next days will eventually confirm it).

claudia-lat, 2019-02-27:
Since this investigation has been done, let's close it.

closed 2019-02-27 by claudia-lat