robotology/wearables #28: Discuss if to maintain RPC support

lrapetti, 2019-02-25:
Following discussion at https://github.com/robotology-playground/wearables/pull/24#discussion_r255020140, we have to discuss if we wan to maintaing `RPC` support.

Currently, `RPC` port is used only if explicitly set to `useRPC=true`.

lrapetti, 2020-11-24:
Since there seems to be no specific need or reason for maintaining this RPC port (as discussed in https://github.com/robotology/wearables/pull/24#discussion_r255020140), I would proceed removing it if there is no-one against it.

cc @Yeshasvitvs @diegoferigo

claudia-lat, 2020-12-01:
@lrapetti understand with @Yeshasvitvs if we can remove the RPC.  Actually it can be useful for PAEXO activity.

claudia-lat, 2020-12-07:
@Yeshasvitvs did you check this?

Yeshasvitvs, 2020-12-09:
@claudia-lat Not yet, we are working on the backend of `iFeelDriver` to use the command communication pipeline first. We will address this check soon after that.

lrapetti, 2021-10-14:
At the end the RPC was not used even for wearable commands, so we can proceed removing it.