robotology/wearables #30: Wearable producer does Not stream data with network instability

kouroshD, 2019-04-03:
During the last experiments for the teleoperation, we were running HDE on a MAC machine which was connected through WiFi to yarpserver as mentioned [#116](https://github.com/dic-iit/lab-organisation/issues/116), and the connection was unstable. 
The problems we were observing were: 

* We should try several times to connect a source port in MAC to a destination port in iCub-head.
* After running the experiment for some time, the controller was working (we could walk with the robot) but we could not teleoperate the upper body of the robot. After plotting, we saw the reference values for the joint torques were fixed with small oscillations.

Investigating more, we reached to the point that the data coming from the HDE is fixed. After talking with @diegoferigo  and talking with @Yeshasvitvs , we ran a number of experiments to simulate the instability of the network. If the connection is lost between wearable (Windows machine connected through LAN) and IWearRemapper (on MAC, connected by WiFi), still the IWearRemapper exposes the last received data (from the wearable producer) to the human state provider, and therefore the human state provider continues running and stream joint pose through the yarp network ports.
As @diegoferigo suggested we check for the yarpnetwork now in IWearRemapper as an optional solution to this problem [here](https://github.com/kouroshD/wearables/blob/feature/ICub-device-impK/devices/IWearRemapper/src/IWearRemapper.cpp#L376).  

After running again the wearables and HDE, we observed the following behaviors:
* If we stop the HDE (MacOS) (ctrl+C), the wearables (Winodws machine) continue to stream data as expected. It makes new tmp ports and write to them. We can access to the streamed data in different machine.

* If we change the Mac connected network (manually connect to IIT network) so that the Mac Machine does not access to yarpserver, we see following behaviors:
    - The Wearables in Windows Machine stop streaming data (even in Windows we can not read data from Wearables), therefore we need to restart the wearables and calibrate again the XSens suit.
    - If we reconnect to icub-green  network, HDE still runs as before and streams fixed value even if there is a check the port connections [here](https://github.com/kouroshD/wearables/blob/feature/ICub-device-impK/devices/IWearRemapper/src/IWearRemapper.cpp#L376).

    - We did not define the sampling time in HDE, therfore it should be its default value. At the begining, when HDE is running correctly,  every `0.5-1` sec it calls the `IWearRemapper::run()`, but when the network is broken (and even after the reconnection to icub-green) for 3-5 times it calls the `run` method with the same rate, but later I will call the method every `20-60`seconds.
  
@Yeshasvitvs Feel free to comment and add if sth is missing or inaccurate!
CC @DanielePucci  @diegoferigo  @Yeshasvitvs  @diegoferigo

diegoferigo, 2019-04-04:
> The Wearables in Windows Machine stop streaming data (even in Windows we can not read data from Wearables), therefore we need to restart the wearables and calibrate again the XSens suit.

This is weird. So if you disconnect your pc (which is the consumer of the data) from the windows machine (which is the producer), then you cannot read the data neither from the producer?

> If we reconnect to icub-green network, HDE still runs as before and streams fixed value even if there is a check the port connections here.

As we discussed f2f, this can be due to the fact the new data is not arriving, and IK still tries to find a solution with the last sample received. You told me that this data slightly varies and I suspect this is due to the IK solution that is not exactly the same at every iteration.

> We did not define the sampling time in HDE, therfore it should be its default value.

I don't get what you mean here, all devices of HDE implement `PeriodThread` and they run typically at 100Hz, with the exception of `HumanStateProvider`.

> At the begining, when HDE is running correctly, every 0.5-1 sec it calls the IWearRemapper::run()

This should not be called by classes attached to the remapper. `run()` is the implementation of `PeriodicThread`. Since the computation of the remapper is triggered by a callback on the input ports, the `run()` is a dummy method linked to a thread with the only purpose to keep the object alive.

kouroshD, 2019-04-04:
> This is weird. So if you disconnect your pc (which is the consumer of the data) from the windows machine (which is the producer), then you cannot read the data neither from the producer?

Yes, it is weird, but that is what we have observed.  cc @Yeshasvitvs 

> As we discussed f2f, this can be due to the fact the new data is not arriving, and IK still tries to find a solution with the last sample received. You told me that this data slightly varies and I suspect this is due to the IK solution that is not exactly the same at every iteration.

Yes, that is my impression as well.

> I don't get what you mean here, all devices of HDE implement PeriodThread and they run typically at 100Hz, with the exception of HumanStateProvider.

I meant, we did not define its period in the application description file(xml file), therefore its period should be the default one. This means that `run` method of the IWearReampper should be called with 100Hz, am I right?

> This should not be called by classes attached to the remapper. run() is the implementation of PeriodicThread. Since the computation of the remapper is triggered by a callback on the input ports, the run() is a dummy method linked to a thread with the only purpose to keep the object alive.

Thank you for correction, yes I remember from our talk! The `run()` method should be called every `period` sec even if it is dummy, am I right?

diegoferigo, 2019-04-04:
> I meant, we did not define its period in the application description file(xml file), therefore its period should be the default one. This means that run method of the IWearReampper should be called with 100Hz, am I right?

No, it is called at 1 Hz. Since the method used to be dummy, this rate is hardcoded:

https://github.com/robotology-playground/wearables/blob/04dd8b2a31aa30e7a2ed3e8e7e9532af0e463ccc/devices/IWearRemapper/src/IWearRemapper.cpp#L112-L115

Yeshasvitvs, 2019-05-06:
@kouroshD please update the relevant details on the changes made to address this issue.

kouroshD, 2019-05-08:
I opened a [PR here](https://github.com/robotology-playground/wearables/pull/38)  to merge what we did to face the network instability.
Additionally, we increased the number of LAN ports, so that we do not have the instability. By closing the PR, we can close this issue.

kouroshD, 2019-12-04:
Since we closed the PR, I close this issue. Now, we use the lan networks and we do not have any more the problem with the network.

closed 2019-12-04 by kouroshD