robotology/wearables #31: Implement VirtualJointKinSensor in wearables

Yeshasvitvs, 2019-04-03:
Following the discussion on [wrench source transformation](https://github.com/dic-iit/component_andy/issues/153#issuecomment-479478038), we decided to implement a new sensor type called `VirtualJointKinSensor` that will represent a single joint sensor which provides `(position velocity and acceleration)` of a joint. An example use case of such a sensor can be seen in the case of `ICub wearable device` that will stream the joint quantities of icub as wearable data. 

The sensor is implemented in https://github.com/robotology-playground/wearables/pull/25/commits/25f9fffcce19e1050ba2f61d27ca1a260a1099b6

Current wearable devices are updated with the new sensor in https://github.com/robotology-playground/wearables/pull/25/commits/4904032ee72024627f97af82afbaba973e3d7ae0

**NOTE:** `XsensSuit` wearable device uses the variable names without the word `Spherical`. This may be a bit mis-leading with the new addition of `VirtualJointKinSensor`. I did not update all the variable names at this moment but this task can be put in todo.

A new issue will be opened for enhancing the `ICub wearable device` with the implementation of `VirtualLinkSensor`

Yeshasvitvs, 2019-04-03:
Following the Implementation of `VirtualJointKinSensor` in `ICub` wearable device https://github.com/robotology-playground/wearables/issues/32, I think we can close this issue.

closed 2019-04-03 by Yeshasvitvs

diegoferigo, 2019-04-04:
I dropped few comments here and there in your code.