robotology/wearables #36: Test two xsens suits together

Yeshasvitvs, 2019-04-23:
Following the testing of the new xsens awinda suit, in this issue we will track the details related using two xsens suits together on the same machine. Ideally, we need to have two [configuration files](https://github.com/robotology-playground/wearables/blob/master/conf/Xsens/xsens_windows_all_wrappers.xml) with the `xsens suit` device [name](https://github.com/robotology-playground/wearables/blob/47962e176793ee3e5b76b707e1e408b1260eb6ba/conf/Xsens/xsens_windows_all_wrappers.xml#L5) set differently for the **xsens link** (wired suit) and **xsens awinda** (wireless suit). Also, all the port of the wrappers should be updated with a prefix accordingly. @kouroshD @lrapetti and I will test the two suits together today

CC @DanielePucci @diegoferigo @claudia-lat

DanielePucci, 2019-04-23:
@Yeshasvitvs thanks for this, if it not too much work for you. Now more important the pHRI

lrapetti, 2019-04-23:
We tested the two xsens suit together by launching two devices with two different prefix, and we were able to run them, calibrate and read the data from the two suits running at the same time.

The only thing left, it is the fact that we are not able to select which specific hardware we want to connect to with a `xsens-yarp` device when both the suits are on. The current solution to that is:
- turn on one suit and run the first device
- wait until the device is connected
- turn on the second suit and run the second device

Yeshasvitvs, 2019-04-23:
The new configuration files are added in https://github.com/robotology-playground/wearables/commit/ba85e3f9dced710b0535ebe5506c8a23c3d4271d

we can close this issue

closed 2019-04-23 by Yeshasvitvs