robotology/wearables #37: Update termination procedure of wearable components

Yeshasvitvs, 2019-05-06:
Following the [Investigate and Fix segmentation fault on HDE termination](https://github.com/robotology/human-dynamics-estimation/issues/109) issue, I noticed that the termination procedure for the wearable components is also incorrect. In this issue we track the details on fixing the termination procedure for wearable components. 

@lrapetti @diegoferigo @traversaro @DanielePucci

DanielePucci, 2019-05-28:
@Yeshasvitvs  please open a iCub PR.

DanielePucci, 2019-07-22:
@Yeshasvitvs could you please update and detail this issue?

Yeshasvitvs, 2019-07-22:
The termination process is fine on linux machine but we noticed that on mac machine it is not clean. The last test we did was on @lrapetti machine. @lrapetti can you please update this issue if you tried to double check this patch further. Thanks.

DanielePucci, 2019-07-29:
It needs to be teste on Mac @lrapetti

lrapetti, 2019-08-05:
I tested the `master` and terminate correctly, while I get segmentation fault on termination for `feature/ICub-dev-impl`

Yeshasvitvs, 2019-08-05:
@lrapetti Please test the branch https://github.com/robotology/wearables/commit/575c6342dfab709d6af09f334043aa2cebfca4d1

Once it is tested we can merge to `master` and then proceed to also merge `feature/ICub-dev-impl` to `master`

lrapetti, 2019-09-17:
New PR at https://github.com/robotology/wearables/pull/50

Yeshasvitvs, 2019-09-18:
The PR #50 is merged, closing this issue.

closed 2019-09-18 by Yeshasvitvs