robotology/wearables #39: Test wearables yarpmanager application

Yeshasvitvs, 2019-05-22:
Following the creation of yarpmanager application for wearables https://github.com/dic-iit/component_andy/issues/195 in this issue we will test if all the components can be launched correctly.

CC @lrapetti

Yeshasvitvs, 2019-07-25:
@lrapetti we should test this as early as possible and close this issue. It will help to launch the demos directly using yarpmanager and reduces bus factor.

lrapetti, 2019-07-30:
I was testing the `yarpmanager` application on Windows laptop using the [`Wearables-yarpmanager.ini`](https://github.com/robotology/wearables/blob/feature/ICub-device-impl/app/Wearables-yarpmanager.ini) file in the branch `ICub-device-impl` but I noticed that the files are not installed and the applications get stuck as they try to find the files.

Yeshasvitvs, 2019-07-30:
@lrapetti HDE laptop and icub30 laptop may have local changes on wearables and HDE which I had to do for pHRI rviz visualization. Please double check this when trying to run anything. If possible stash the changes as we still have to get rviz visualization.

Yeshasvitvs, 2020-01-27:
@lrapetti any update on this ?

Yeshasvitvs, 2021-06-15:
@lrapetti may be we can close this issue ?

lrapetti, 2021-06-16:
> @lrapetti may be we can close this issue ?

I think so, we already used some of those application

closed 2021-06-16 by lrapetti