robotology/wearables #41: Move to Robotology?

kouroshD, 2019-06-03:
Since we are planning to add HDE to robotology-superbuild, I think it is a good occasion to move this repo to robotology, since it is stable as well.
What do you think and suggest @traversaro ?

cc @DanielePucci  @Yeshasvitvs  @lrapetti  @claudia-lat @diegoferigo

traversaro, 2019-06-03:
I totally agree!

DanielePucci, 2019-06-03:
**go!**

diegoferigo, 2019-06-03:
Double thumbs up!

kouroshD, 2019-06-04:
Since I do not have the permission at organizational level, I cannot proceed further for this issue.
Also, I suggest @Yeshasvitvs  follow this issue, since he is supposed to be the maintainer of the `Human_Dynamics` profile.

@traversaro  @diegoferigo  What do you suggest? should we move the repo to the other organization or we should fork it?

traversaro, 2019-06-04:
I can move it, thanks!

traversaro, 2019-06-04:
Done.

closed 2019-06-04 by traversaro