robotology/wearables #46: WearableData thrift message not working in YARP 3.2

lrapetti, 2019-07-08:
Currently, the bug described in https://github.com/robotology/yarp/issues/2054 prevents the usage of `YARP 3.2` with `wearable` due to the impossibility of communicating trough WearableData messages.

For this reason, it is suggested not to update YARP until the issue is solved. Unfortunately, installing all the required software in a new machine rolling back to an older version of YARP is not really straightforward at the moment since it is not possible to use the [`robotology-superbuild`](https://github.com/robotology/robotology-superbuild) (or at least the `master`) because of many recent changes that took place (such as [Robot Testing Framework 2.0 release](https://github.com/robotology/QA/issues/351)).

CC @Yeshasvitvs @kouroshD @diegoferigo @traversaro

diegoferigo, 2019-07-08:
If this problem persists for some while I would recommend to create a `ProjectsTagsAndy.cmake` similar to [`ProjectsTagsDevel.cmake`](https://github.com/robotology/robotology-superbuild/blob/master/cmake/ProjectsTagsDevel.cmake) containing the commits pre-yarp 3.2, and then load the new file in the [main CMakeLists.txt](https://github.com/robotology/robotology-superbuild/blob/f3ec55e5d75f540c988083e52808ac1a60005f06/CMakeLists.txt#L83-L87).

lrapetti, 2019-07-08:
> If this problem persists for some while I would recommend to create a `ProjectsTagsAndy.cmake` similar to [...]

I think it is also required to roll back in a previous commit of the `robotology-superbuild` because of https://github.com/robotology/robotology-superbuild/commit/3b969a8c2336d053c216480680b7b45d0cb904f4

lrapetti, 2019-09-19:
https://github.com/robotology/yarp/commit/235871e75bc9abbd821ee97eea2880b566f82df3 fixed the problem.

closed 2019-09-19 by lrapetti