robotology/wearables #48: Incorporate FTBox inside wearables

Yeshasvitvs, 2019-08-12:
The recently designed FTBox is to be used with pHRI and pRRI experiments. The box has four FT sensors one at each of the handles that will be help either by the human/robot. In the case of FTShoes, we use the [ftshoe](https://github.com/robotology/forcetorque-yarp-devices/tree/master/ftShoe) that handles data transformation from the front and rear sensors to a frame attached to the feet of the human. Unlike the FTShoes, in the case of FTBox we will need to read the data from individual sensors without any need to fuse them to be expressed at a single frame (at least for the time being). So, in order to use the FTBox it will be convenient to launch it as a wearable device. Fortunately, the [IAnalogSensorToIWear](https://github.com/robotology/wearables/tree/master/devices/IAnalogSensorToIWear) wearable device makes it possible easily. The main components to incorporate FTBox as part of wearables repository is:

- Launch `canBusAnalogSensor` for each FTSensor. This is [similar](https://github.com/robotology/wearables/blob/feature/ICub-device-impl/app/xml/FTShoeLeftWearableDevice.xml#L5) to the config file used for launching FTShoes as a wearable device.
- Launch `analogServer` for each FTSensor and attach the `canBusAnalogSensor` device from the previous step to stream the data in yarp ports. This is [similar](https://github.com/robotology/wearables/blob/42eba1b291ebfda8a27629c4a999cf1225bdca30/app/xml/FTShoeLeftWearableDevice.xml#L100) to the config file used for launching FTShoes as a wearable device.
- Launch `ianalogsensor_to_iwear` wearable device for each FTSensor and attach the `canBusAnalogSensor` device from the first step. This is [similar](https://github.com/robotology/wearables/blob/feature/ICub-device-impl/app/xml/FTShoeLeftWearableDevice.xml#L121) to the config file used for launching FTShoes as a wearable device.
- Finally, launch `iwear_wrapper` and attach the `ianalogsensor_to_iwear` from the previous step to stream the FTSensor data as wearable data. This is [similar](https://github.com/robotology/wearables/blob/42eba1b291ebfda8a27629c4a999cf1225bdca30/app/xml/FTShoeLeftWearableDevice.xml#L136) to the config file used for launching FTShoes as a wearable device.

One key decision we need to make is the naming of the FT sensors. In the case of FTShoes it was clear as `FTShoeLeft_Front` and `FTShoeLeft_Rear` or `FTShoeRight_Front` and `FTShoeRight_Rear`. In the case of the FTBox this is a bit unclear for me to have a clear naming convention. As discussed with @lrapetti as possible naming would be as following, with the numbers and the cable covering hood as shown in the figure.

[image]

@claudia-lat @DanielePucci

Yeshasvitvs, 2019-08-12:
The config file is added in https://github.com/robotology/wearables/commit/0477218f65c36956a6e161cb67b077230cef393e

DanielePucci, 2019-08-15:
@Yeshasvitvs thanks a lot for this.

> One key decision we need to make is the naming of the FT sensors.

Another possibility is the following. If we assume a human-robot collaboration scenario, there will be a human and a robot carrying the box. So, assuming also that the box is *face-up*, we may call the sensors in this way:
* `FTBoxLeft_HumanHand`
* `FTBoxRight_HumanHand`
* `FTBoxLeft_RobotHand`
* `FTBoxRight_RobotHand`

What do you think @Yeshasvitvs @lrapetti @claudia-lat ?

lrapetti, 2019-08-16:
> * `FTBoxLeft_HumanHand`
> * `FTBoxRight_HumanHand`
> * `FTBoxLeft_RobotHand`
> * `FTBoxRight_RobotHand`
> 
> What do you think @Yeshasvitvs @lrapetti @claudia-lat ?

Given that we are not planning to use the box only in human-robot but eventually also robot-robot and human-human, I think `subject1` and `subject2` can work as well. I think in any case we can mark the name of the sensors/sides on the box in order to avoid confusion.

Yeshasvitvs, 2019-08-19:
@DanielePucci we ran into a similar problem on HDE too. As we are advancing to doing both pHRI pRRI use of `HumanHand` and `RobotHand` gets confusing quickly. I agree with @lrapetti to add the sensor numbering (Similar to that of FTShoes).

claudia-lat, 2019-09-23:
@Yeshasvitvs add a comment on this issue.

Yeshasvitvs, 2019-09-23:
As discussed with @DanielePucci we will be proceeding to use the wireless nodes to stream the data from the FT sensors of the FTBox. So, as the FTBox contains the same number of sensors as the FTShoes, we can directly use the [FTNode](https://github.com/Yeshasvitvs/forcetorque-yarp-devices/tree/feature/ftNode) that is recently tested with the FTShoes. 

Under this architecture, the data flow will be
- Wireless receiver streaming the data of the four FT sensors of the FTBox to a serial port
- A yarp serial device will be opened to read the incoming data in the serial port
- FTNode device attaches to the serial device and streams the FT data through a single `IAnalogServer` interface with 24 channels (4 FT sensors x 6 channels per FT sensor)
- Finally, a `AnalogServer` device will be opened and attaches to the FTNode and yarp ports or ros topics for each sensor data can be streamed to yarp network. Check https://github.com/robotology/human-dynamics-estimation/blob/devel/conf/xml/Human.xml#L238 for example

So, considering the FTBox, the main points to do is to write the correct configuration file and nothing more is needed at this stage.

@lrapetti @claudia-lat @DanielePucci 

Closing this issue

closed 2019-09-23 by Yeshasvitvs