robotology/wearables #51: Investigate using two IAnalogToIWear devices in the same xml configuration file

Yeshasvitvs, 2019-09-18:
Currently, the FTShoes are launched using two separate xml configuration files, one for the [left](https://github.com/robotology/wearables/blob/master/app/xml/FTShoeLeftWearableDevice.xml) and one for the [right](https://github.com/robotology/wearables/blob/master/app/xml/FTShoeRightWearableDevice.xml). Ideally, we should be able have a single configuration file to launch both the shoes as done in e.g. https://github.com/robotology/forcetorque-yarp-devices/blob/master/ftShoe/conf/ftshoes_yarprobotinterface_PRO.02.xml. The only new device driver that is added is the [IAnalogSensorToIWear](https://github.com/robotology/wearables/tree/master/devices/IAnalogSensorToIWear) to parse the data to wearable data. Putting two `IAnalogSensorToIWear` devices (one for each shoe) in the same configuration file is throwing some error (will update the issue with the error soon) when launched with `yarprobotinterface`. In this issue we investigate the problem and ensure we can use a single configuration file with two `IAnalogSensorToIWear`. 

Solving this issue will enable use to use the both the FTShoes through wireless node. 

@lrapetti @claudia-lat @kouroshD @diegoferigo @traversaro @DanielePucci

lrapetti, 2019-09-23:
When two `ianalogsensor_to_iwear` are in the same file like
```
<device type="ianalogsensor_to_iwear" name="FTShoeLeftIAnalogSensorToIWear">
    <param name="sensorName">FTShoeLeftFTSensors</param>
    <param name="wearableName">FTShoeLeft</param>
    <param name="numberOfChannels">6</param>
    <param name="channelOffset">0</param>
    <param name="wearableSensorType">ForceTorque6DSensor</param>
    <param name="getGroundReactionFT">true</param>
    <action phase="startup" level="5" type="attach">
        <paramlist name="networks">
            <elem name="FTShoeLeftIAnalogSensorToIWearLabel">ftShoe_l</elem>
        </paramlist>
    </action>
    <action phase="shutdown" level="5" type="detach"/>
</device>
<device type="ianalogsensor_to_iwear" name="FTShoeRightIAnalogSensorToIWear">
    <param name="sensorName">FTShoeRightFTSensors</param>
    <param name="wearableName">FTShoeRight</param>
    <param name="numberOfChannels">6</param>
    <param name="channelOffset">0</param>
    <param name="wearableSensorType">ForceTorque6DSensor</param>
    <param name="getGroundReactionFT">true</param>
    <action phase="startup" level="5" type="attach">
        <paramlist name="networks">
            <elem name="FTShoeRightIAnalogSensorToIWearLabel">ftShoe_r</elem>
        </paramlist>
    </action>
    <action phase="shutdown" level="5" type="detach"/>
</device>
```

We are getting the following error:
```
[INFO]IAnalogSensorToIWear : *** ====================
[INFO]IAnalogSensorToIWear : *** Sensor name        : FTShoeLeftFTSensors
[INFO]IAnalogSensorToIWear : *** Sensor Type        : ForceTorque6DSensor
[INFO]IAnalogSensorToIWear : *** Wearable name      : FTShoeLeft
[INFO]IAnalogSensorToIWear : *** Number of channels : 6
[INFO]IAnalogSensorToIWear : *** Channel offset     : 0
[INFO]IAnalogSensorToIWear : *** Ground reaction FT : true
[INFO]IAnalogSensorToIWear : *** ====================
[INFO]created device <ianalogsensor_to_iwear>. See C++ class wearable::devices::IAnalogSensorToIWear for documentation.
[INFO]IAnalogSensorToIWear : *** ====================
[INFO]IAnalogSensorToIWear : *** Sensor name        : FTShoeRightFTSensors
[INFO]IAnalogSensorToIWear : *** Sensor Type        : ForceTorque6DSensor
[INFO]IAnalogSensorToIWear : *** Wearable name      : FTShoeRight
[INFO]IAnalogSensorToIWear : *** Number of channels : 6
[INFO]IAnalogSensorToIWear : *** Channel offset     : 0
[INFO]IAnalogSensorToIWear : *** Ground reaction FT : true
[INFO]IAnalogSensorToIWear : *** ====================
[INFO]created device <ianalogsensor_to_iwear>. See C++ class wearable::devices::IAnalogSensorToIWear for documentation.
||| finding paths [plugins]
[INFO]created device <iwear_wrapper>. See C++ class wearable::wrappers::IWearWrapper for documentation.
[INFO]Entering action level 5 of phase startup
yarp: Port /ft/ftShoe_Left/analog:o active at tcp://10.0.0.121:10006/
yarp: Port /ft/ftShoe_Right/analog:o active at tcp://10.0.0.121:10014/
||| finding paths [plugins]
yarp: Port /ft/ftShoe_Left_Front/analog:o active at tcp://10.0.0.121:10007/
yarp: Port /ft/ftShoe_Left_Rear/analog:o active at tcp://10.0.0.121:10008/
yarp: Port /ft/ftShoe_Right_Front/analog:o active at tcp://10.0.0.121:10015/
yarp: Port /ft/ftShoe_Right_Rear/analog:o active at tcp://10.0.0.121:10016/
[ERROR]IAnalogSensorToIWear : Failed to view the IAnalogSensor interface from the PolyDriver
[ERROR]Device FTShoeRightIAnalogSensorToIWear cannot execute attach
[ERROR]Cannot run attach action on device FTShoeRightIAnalogSensorToIWear
```

lrapetti, 2019-09-23:
The device seems to be failing when checking the existence of `handler.interface` pointer (at https://github.com/robotology/wearables/blob/master/devices/IAnalogSensorToIWear/src/IAnalogSensorToIWear.cpp#L478) for the second `iAnalogSensorToIWear`. This may be caused by some static variable shared across the two instances of the device.

The first thing to notice is that the `handler` class is defined as [`static`](https://github.com/robotology/wearables/blob/master/devices/IAnalogSensorToIWear/src/IAnalogSensorToIWear.cpp#L28), while it shouldn't. However, even after removing the `static` the same problem appear. As suggested by @diegoferigo, I have tried to verify the address of the `handler` pointer, and it seems it is the same among the two `iAnalogSensorToIWear` instances. This confirm the problem is due to the existence of static pointer shared among the two.

lrapetti, 2019-09-23:
I tried to add the following debug code in the `attach()` function for checking the address of the pointers:
```
    if (!poly) {
        yError() << LogPrefix << "Passed PolyDriver is nullptr";
        return false;
    }
	
	yInfo() << "POLIDRIVER ADDRESS: " << &poly;
	yInfo() << "HANDLER ADDRESS BEFORE: " << &handler;

    if (handler.interface || !poly->view(handler.interface) || !handler.interface) {
        yError() << LogPrefix << "Failed to view the IAnalogSensor interface from the PolyDriver";
        return false;
    }
	
	yInfo() << "HANDLER ADDRESS AFTER: " << &handler;
    ....
```

and the output is the following;
```
[INFO]POLIDRIVER ADDRESS:  0000006BA14FE0F8
[INFO]HANDLER ADDRESS BEFORE:  00007FFA517219D0
[INFO]HANDLER ADDRESS AFTER:  00007FFA517219D0
[INFO]POLIDRIVER ADDRESS:  0000006BA14FE0F8
[INFO]HANDLER ADDRESS BEFORE:  00007FFA517219D0
[ERROR]IAnalogSensorToIWear : Failed to view the IAnalogSensor interface from the PolyDriver
```

so the problem is also for the polydriver pointer I think.

diegoferigo, 2019-09-23:
Yes it seems that the device driver (the shoe device implementing `IAnalogSensor`) passed to the two `IAnalogSensorToIWear` is the same object :thinking:

What's strange is that also in the first attach the `handler address before != nullptr` and the if shuold fail. Can you check if the pointer is initialized to `nullptr` when it is declared?

diegoferigo, 2019-09-23:
Anyway, as it is written, if `handler.interface != nullptr`, given the or, the polidriver proceeds to attach anyway. It should instead fail.

Try with:

```cpp
if (!(!handler.interface && poly->view(handler.interface) && handler.interface)) {
    yError() << LogPrefix << "Failed to view the IAnalogSensor interface from the PolyDriver";
    return false;
}
```

In this way if the first statement fails, the execution does not continue. Btw this is just a nitpick and it should not solve the original problem.

lrapetti, 2019-09-23:
> Can you check if the pointer is initialized to `nullptr` when it is declared?

The `handler` is not a pointer, but `handler.interface` is initialized [as `nullpointer`](https://github.com/robotology/wearables/blob/master/devices/IAnalogSensorToIWear/src/IAnalogSensorToIWear.cpp#L32).

lrapetti, 2019-09-23:
I have tried to remove the check during the attach step, and the two devices where running both being attached to the analog interface of the first shoes (streaming the same wearable data). This confirm that there is  problem also with the input `polyDriver`.

lrapetti, 2019-09-23:
In order to have an example of a similar device that is working, I printed the same address for the [`IWearWrapper`](https://github.com/robotology/wearables/blob/master/wrappers/IWear/src/IWearWrapper.cpp#L470). The output is the following:
```
[INFO]Entering action level 5 of phase startup 
[INFO]XXXXXXXXXXXXXXXXXX poly driver address:  0x7ffeeda88880 
[INFO]XXXXXXXXXXXXXXXXXX iWear address before:  0x7fda19517f18 
[INFO]XXXXXXXXXXXXXXXXXX iWear address after:  0x7fda19517f18 
yarp: Port /WearableData/data:o active at tcp://10.240.2.16:10010/
[DEBUG]IWearWrapper : attach() successful 
[INFO]XXXXXXXXXXXXXXXXXX poly driver address:  0x7ffeeda88880 
[INFO]XXXXXXXXXXXXXXXXXX iWear address before:  0x7fda1951e028 
[INFO]XXXXXXXXXXXXXXXXXX iWear address after:  0x7fda1951e028 
yarp: Port /WearableData2/data:o active at tcp://10.240.2.16:10012/
```

The interesting thing here is that the `polyDriver` also in this case has the same address. This means that the pointer for the polydriver is somehow overwritten at every attach call. On the other side, the `iWear` interface in this case has a different address for each device instante. The main difference is that in `IAnalogSensoroToIWear` the interface is not defined as part of the object, but it is declared as global in the `.h`. I will try to add it to the `pImpl` to check if this solve the issue.

diegoferigo, 2019-09-23:
I suspect that the object passed to the attach method is not a single polydriver but something else, like a list of polydrivers. I'm not very familiar with this YARP corner tho. Maybe @traversaro can provide further information.

Anyways, the double attach to the same device is quite curious :)

traversaro, 2019-09-24:
> I suspect that the object passed to the attach method is not a single polydriver but something else, like a list of polydrivers. I'm not very familiar with this YARP corner tho. Maybe @traversaro can provide further information.

As far as I can see from the code in https://github.com/robotology/wearables/issues/51#issuecomment-534095414, @lrapetti was not printing the  address **contained in** the `yarp::dev::PolyDriver* poly` (i.e. `std::cerr << poly`)  but rather **the  address of** the local argument poly itself (i.e. `std::cerr << &poly`), that is make sense that it is constant if the call stack (i.e. the chaing of the function that call the `attach` method) does not change.

diegoferigo, 2019-09-24:
Ow poly is a pointer, not a (referenced) object. I lost the sight of the function signature, my mistake.

lrapetti, 2019-09-27:
Solved by #52

closed 2019-09-27 by lrapetti