robotology/wearables #56: CI failing on MacOS

lrapetti, 2019-10-09:
See https://travis-ci.org/robotology/wearables/builds/595555081?utm_medium=notification&utm_source=email. The Travis CI seems to be broken on MacOS, probably after some update in YARP.

traversaro, 2019-10-23:
I guess this is related to https://github.com/robotology/yarp/issues/2044 , even if I am not sure why it fails only in some specific case.

traversaro, 2019-10-23:
Other possibly related issues: https://github.com/robotology/yarp/pull/2110 .

traversaro, 2019-10-23:
Actually I think that the regression was introduced by https://github.com/robotology/yarp/pull/2102 .

lrapetti, 2019-10-24:
As an initial step I am restoring the CI to the standard deps branches (https://github.com/robotology/wearables/pull/58) so that the CI for the master should be fixed. I will then investigate further the compatibility with YARP `devel`.

lrapetti, 2019-10-25:
> Actually I think that the regression was introduced by [robotology/yarp#2102](https://github.com/robotology/yarp/pull/2102) , and the reason why it is not failing in Travis/Linux is that Linux is using an old version of YARP due to the docker caching.

I confirm the regression in introduced by this commit

traversaro, 2019-11-06:
As the upstream issue is not really progressing, I would strongly suggest to have some sort of local workaround that fixes the build for the time being, for example by modifying the include to be `include "WearableData.h"`, and modify the CMake script in https://github.com/robotology/wearables/blob/f52a9e1206e511a6a782add69ba4d3880ed71ce9/msgs/CMakeLists.txt#L34 to have in `WEARABLEDATA_INCLUDE_DIRS` the necessary directories to be compatible both with YARP 3.2 and the future YARP 3.3 .

lrapetti, 2019-11-07:
> I would strongly suggest to have some sort of local workaround that fixes the build for the time being, for example by modifying the include to be `include "WearableData.h"`

I tried this fix, the problem is that the same problem is raised for the include of the generated `WearableData.h`:
```
/Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/wearables/msgs/thrift/wearable/msg/WearableData.h:19:10: fatal error: 
      'thrift/wearable/msg/Accelerometer.h' file not found
#include <thrift/wearable/msg/Accelerometer.h>
```
In fact in its header it includes all the other thrift messages:
```
#include <thrift/wearable/msg/FreeBodyAccelerationSensor.h>
#include <thrift/wearable/msg/Gyroscope.h>
#include <thrift/wearable/msg/Magnetometer.h>
#include <thrift/wearable/msg/OrientationSensor.h>
#include <thrift/wearable/msg/PoseSensor.h>
#include <thrift/wearable/msg/PositionSensor.h>
#include <thrift/wearable/msg/SkinSensor.h>
#include <thrift/wearable/msg/TemperatureSensor.h>
#include <thrift/wearable/msg/Torque3DSensor.h>
#include <thrift/wearable/msg/VirtualJointKinSensor.h>
#include <thrift/wearable/msg/VirtualLinkKinSensor.h>
#include <thrift/wearable/msg/VirtualSphericalJointKinSensor.h>
```

> and modify the CMake script in
> 
> https://github.com/robotology/wearables/blob/f52a9e1206e511a6a782add69ba4d3880ed71ce9/msgs/CMakeLists.txt#L34
> 
> to have in `WEARABLEDATA_INCLUDE_DIRS` the necessary directories to be compatible both with YARP 3.2 and the future YARP 3.3 .

Here to you mean for example to add also the parent directory of the `thrift` directory?

traversaro, 2019-11-07:
> Here to you mean for example to add also the parent directory of the `thrift` directory?

Exactly. Just to understand, with YARP 3.3 , the include headers contain the `thrift` directory, but then the generated file is not in the `thrift` directory? This is even worse, for better visibility I suggest to open a new YARP issue instead of just commenting an old closed PR. 
An alternative for the problem may be to move the `*.thrift` files outside of the `thrift` directory, perhaps that will work with both YARP 3.2 and 3.3 .

lrapetti, 2019-11-11:
> for better visibility I suggest to open a new YARP issue instead of just commenting an old closed PR.

new PR opened at https://github.com/robotology/yarp/issues/2118.

lrapetti, 2019-11-11:
> Exactly. Just to understand, with YARP 3.3 , the include headers contain the `thrift` directory, but then the generated file is not in the `thrift` directory?

No, what happen is that all the generated files (e.g. `Accelerometer.cpp`, `Accelerometer.h`, `WearableData.h`, `WearableData.cpp`) are in the same folder `build/robotology/wearable/msgs/thrift/wearable/msg`, but still the `WearableData.h` cannot find the headers of the other generated messages with the error message above:
```
/Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/wearables/msgs/thrift/wearable/msg/WearableData.h:19:10: fatal error: 
      'thrift/wearable/msg/Accelerometer.h' file not found
#include <thrift/wearable/msg/Accelerometer.h>
```

> An alternative for the problem may be to move the `*.thrift` files outside of the `thrift` directory, perhaps that will work with both YARP 3.2 and 3.3 .

Same happen in this situation, and what change is just that the folder in which they are is: `build/robotology/wearable/msgs/wearable/msg`, and I am getting the error:
```
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp:12:
/Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/wearables/msgs/wearable/msg/WearableData.h:19:10: fatal error: 'wearable/msg/Accelerometer.h' file not found
#include <wearable/msg/Accelerometer.h>
```

traversaro, 2019-11-11:
Can't you just pass `build/robotology/wearable/msgs` as additonal include directory?

traversaro, 2019-11-11:
Or any directory that is necessary to make sure that everything is correctly included.

lrapetti, 2019-11-12:
I am trying to fix the compilation with YARP devel:
- the first step was to remove the `thrift/` scoping so that we avoid the problem of differeces between master and devel. This was achieved with https://github.com/lrapetti/wearables/commit/d89d6ea6f64e39395713abe888de548e4826fa1b that was tested comipiling in `master`. I can see in case of using master all the generated .h messages have the include without any scoping . In case of using YARP devel however, all of them are generated using `wearable/msg/` scoping, and indeed the following error is raised:
  ```
  In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/wrappers/IWear/src/IWearWrapper.cpp:11:
  /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/wearable/msg/WearableData.h:19:10: fatal error: 'wearable/msg/Accelerometer.h' file not found
  #include <wearable/msg/Accelerometer.h>
  ```
- to avoid that error I tried to add the path to the include directory, as @traversaro suggested, adding `list(APPEND WEARABLEDATA_INCLUDE_DIRS ${CMAKE_CURRENT_BINARY_DIR}/msgs/thrift/wearable/msg)` in https://github.com/lrapetti/wearables/commit/a868e32e21055e407be33e253cc0e428a3b9c2df, and it was able to compile all the messages in YARP devel. However, it results in the following error:
  ```
  /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp:12:10: fatal error: 'WearableData.h' file not found
  #include "WearableData.h"
  ```
  and the same error results when trying to compile with YARP master.

traversaro, 2019-11-12:
Can't we append the directory of the generation of the `WearableData.h` file in the include directories as well, or change the `#include "WearableData.h"` line to have a path compatible with the used include dirs (for example by copying from whatever stile is used in the generated `WearableData.cpp`?).

lrapetti, 2019-11-12:
> Can't we append the directory of the generation of the `WearableData.h` file in the include directories as well

I tried also this, but then it was failing to find `IWearRemapper.h` from `yarp_plugin_iwear_remapper.cpp`. However I will double check.

> or change the #include "WearableData.h" line to have a path compatible with the used include dirs (for example by copying from whatever stile is used in the generated WearableData.cpp?).

The problem is that this path would be different using YARP master or devel

traversaro, 2019-11-12:
> I tried also this, but then it was failing to find `IWearRemapper.h` from the generated `yarp_plugin_iwear_remapper.cpp`. However I will double check.

This seems to be a different problem.

> The problem is that this path would be different using YARP master or devel

Ack, then it is not ideal, but if this is the only viable option you can also have an `#ifdef` from a definition that you can pass from CMake using `target_compile_definitions`  depending on the YARP version.

lrapetti, 2019-11-12:
> Can't we append the directory of the generation of the `WearableData.h` file in the include directories as well

Anyway, I am not understating why I have this problem also in `YARP master` after adding the path for including the generated messages (i.e. after https://github.com/lrapetti/wearables/commit/a868e32e21055e407be33e253cc0e428a3b9c2df), while previously to that commit It is compiling without any error in `master`.

traversaro, 2019-11-12:
> 
> 
> > Can't we append the directory of the generation of the `WearableData.h` file in the include directories as well
> 
> Anyway, I am not understating why I have this problem also in `YARP master` after adding the path for including the generated messages (i.e. after [lrapetti@a868e32](https://github.com/lrapetti/wearables/commit/a868e32e21055e407be33e253cc0e428a3b9c2df)), while previously to that commit It is compiling without any error in `master`.

I am not really sure, probably inspecting directly the command line parameters passed  to the compiler for the compilation unit that is failing (for make via `VERBOSE=1 make`) may provide useful information.

traversaro, 2019-11-12:
@lrapetti let me know if you need any f2f help.

lrapetti, 2019-11-12:
Using verbose I notice something strange, in fact there is a path that is repeated twice: `-I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/msgs/thrift/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/include`. The complete output is:
```
[ 75%] Building CXX object devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o
cd /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/devices/IWearRemapper && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DIWearRemapper_EXPORTS -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/include -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/interfaces/IWear/include -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/msgs/thrift/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/include -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/msgs/thrift/wearable/msg -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/impl/SensorsImpl/include -isystem /Users/lorenzorapetti/Software/robotology-superbuild/build/install/include  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -fPIC   -std=gnu++14 -o CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o -c /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp:12:10: fatal error: 
      'WearableData.h' file not found
#include "WearableData.h"
```

While before [that commit](https://github.com/lrapetti/wearables/commit/a868e32e21055e407be33e253cc0e428a3b9c2df) there is not that wrong path, while there is the correct included path `-I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/include`:

```
[ 75%] Building CXX object devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o
cd /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/devices/IWearRemapper && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DIWearRemapper_EXPORTS -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/include -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/interfaces/IWear/include -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/include -I/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/impl/SensorsImpl/include -isystem /Users/lorenzorapetti/Software/robotology-superbuild/build/install/include  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -fPIC   -std=gnu++14 -o CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o -c /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp
```

traversaro, 2019-11-12:
This is interesting, can you print `WEARABLEDATA_INCLUDE_DIRS` before and after the `list(APPEND ..)`.

lrapetti, 2019-11-12:
> This is interesting, can you print `WEARABLEDATA_INCLUDE_DIRS` before and after the `list(APPEND ..)`.

```
WEARABLEDATA_INCLUDE_DIRS= /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/include
WEARABLEDATA_INCLUDE_DIRS= /Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/include;/Users/lorenzorapetti/Software/robotology-superbuild/robotology/wearables/build/msgs/thrift/msgs/thrift/wearable/msg
```

traversaro, 2019-12-04:
Apparently there has been another regression related to this, see https://github.com/robotology/yarp/issues/2118#issuecomment-561614196 .

lrapetti, 2019-12-17:
With https://github.com/robotology/wearables/pull/71 the new regression should be addressed. Closing the issue.

closed 2019-12-17 by lrapetti