robotology/wearables #57: Is there a branch compatible with YARP unstable branch?

traversaro, 2019-10-20:
**Is your feature request related to a problem? Please describe.**
As described in https://github.com/robotology/robotology-superbuild/issues/101, we would like to setup a CI job that automatically compiles all the unstable branches of the robotology software, to catch as soon as possible any compilation regrcheession. 
We have set it up in https://github.com/robotology/robotology-superbuild/pull/275, but now when compiling all the superbuild with the "unstable" branches (i.e. the `devel` branches) we are facing a compilation issue with Wearables:
~~~
[ 73%] Building CXX object devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/yarp_plugin_iwear_remapper.cpp.o
[ 75%] Building CXX object devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o
/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp:12:10: fatal error: thrift/WearableData.h: No such file or directory
 #include "thrift/WearableData.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/build.make:75: recipe for target 'devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o' failed
make[5]: *** [devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o] Error 1
make[4]: *** [devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/all] Error 2
make[3]: *** [all] Error 2
CMakeFiles/Makefile2:339: recipe for target 'devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/all' failed
Makefile:129: recipe for target 'all' failed
CMakeFiles/wearables.dir/build.make:113: recipe for target 'robotology/wearables/CMakeFiles/YCMStamp/wearables-build' failed
make[2]: *** [robotology/wearables/CMakeFiles/YCMStamp/wearables-build] Error 2
make[1]: *** [CMakeFiles/wearables.dir/all] Error 2
make: *** [all] Error 2
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/wearables.dir/all' failed
Makefile:140: recipe for target 'all' failed
##[error]Process completed with exit code 2.
~~~
It would be great to have at least one branch that is compatible with YARP unstable branch, now `devel` and soon `master` (see https://github.com/robotology/QA/issues/365), as this is the version of YARP tipically used in IIT@Genoa robots. This branch can be master that is kept compatible with both YARP stable and unstable, or a unstable branch. 

The specific compilation problems seems to be similar to https://github.com/robotology/wearables/issues/56 .

**Describe the solution you'd like**
A fix for the specific issue, and then some indication about which branch to use for Werables in the unstable tags of the superbuild (see https://github.com/robotology/robotology-superbuild/pull/275).

lrapetti, 2019-10-20:
@traversaro just to be clear, in this case the "unstable branch" you are mentioning is the devel branch of YARP, and you are trying to compile the `master` branch of `wearable`?
In case I think the problem as you were saying is the same that is present with the `master` branch of YARP and described in https://github.com/robotology/wearables/issues/56.

traversaro, 2019-10-22:
> @traversaro  just to be clear, in this case the "unstable branch" you are mentioning  is the devel branch of YARP

Yes, "unstable" branch of YARP is currently `devel`, even if this is changing soon (see https://github.com/robotology/QA/issues/365). 

> and you are trying to compile the master branch of wearable?

Yes, and I was wondering if I was supposed to use some other branch. 

>  In case I think the problem as you were saying is the same that is present with the master branch of YARP and described in #56.

Interestingly, the CI of the master/master branch in the robotology-superbuild is working, the only one that is failing is the one with YARP devel and wearables master.

lrapetti, 2019-10-23:
> Yes, and I was wondering if I was supposed to use some other branch.

No, `master` is the stable branch of wearable, and it is the one we are mostly using.

traversaro, 2019-10-23:
Ok, this answer the original question, while the thrift include error can be tracked in https://github.com/robotology/wearables/issues/56 .

Yeshasvitvs, 2019-10-24:
Just to summarize the issue:

- Current working/stable branch of `wearables` is the `master` branch, which compiles correctly without any problems with the `master` (stable) branch of `yarp`.

> It would be great to have at least one branch that is compatible with YARP unstable branch, now devel and soon master (see robotology/QA#365), as this is the version of YARP tipically used in IIT@Genoa robots. 

If I understood the comment well, we need to have at least one branch on `wearables` that is compatible with the unstable branch of `yarp`. Currently we use the `master` branch of `wearables`. So, not action needed to here. @traversaro please confirm this. As there is no `devel` branch on `wearables` as the development involved more of feature branches that were quickly merged to the `master` branch. If we need to ensure CI testing for `wearables` using "unstable" branches, we can create a new `devel` branch from the current `master` branch, but in light of the upcoming changes in branch handling https://github.com/robotology/QA/issues/365 I am assuming this is not necessary.

@traversaro Let me know, if we can close this issue

traversaro, 2019-10-24:
Exactly @Yeshasvitvs , if `master` of YARP is meant to be compatible with `devel`/`master` of YARP, no further action is necessary, sorry that I forgot to close the issue.

closed 2019-10-24 by traversaro