robotology/wearables #59: Add another yarprobotinterface application file for calibrating Xsens suit

kouroshD, 2019-11-05:
According to issue:
https://github.com/dic-iit/component_wholebody-teleoperation/issues/194

- For calibrating the XSENS for the retargeting use to the real robot, we have a number of options namely NposeWalk, TposeWalk, Npose, Tpose. With our experience the most reliable one is the one related to NposeWalk. It provides better estimates for the feet related issue. If we use only head or arms maybe Tpose works fine.

  - Solution: We will have two config files for Xsens suit calibration the sake of being fast, so according to the timing or demo, we can go for different calibration.

lrapetti, 2019-11-05:
Morover, I think it may be useful to investigate the possibility of changing the calibration type after launching the device trough the `rpc`

kouroshD, 2019-11-07:
The goal of this issue is to change the calibration type easier, without loosing time. Also running different yarprobotinterface means we should terminate the wearables and run it again. About the rpc, it exists already, only we were not aware of it! Here are the reference to them:

https://github.com/robotology/wearables/blob/aab0b79db716d51094f6f46e80947339e733bc60/msgs/thrift/XsensSuitControlService.thrift#L26

https://github.com/robotology/wearables/blob/aab0b79db716d51094f6f46e80947339e733bc60/wrappers/IXsensMVNControl/src/IXsensMVNControlWrapper.cpp#L33

I have tested that and it works fine, like:
```
calibrateWithType NposeWalk 
calibrateWithType TposeWalk
calibrateWithType Npose
calibrateWithType Tpose
```

So, we can close this issue.

closed 2019-11-07 by kouroshD