robotology/wearables #66: Investigate Xsens calibration with NposeWalk instead of NPoseWalk

Yeshasvitvs, 2019-11-18:
While testing xsens calibration with the wired suit, @kouroshD and I  noticed that the calibration was very bad. On double checking we noticed that the calibration command used through xsens rpc port is `calibrateWithType NposeWalk` instead of `calibrateWithType NPoseWalk`. 

Using the `calibrateWithType NposeWalk` calibration procedure seems to be correct as in it was displaying the standard steps of `Standing in NPose`, `walking front` and then `walking back` and the final result of calibration is shown `ok`. However, the final result of calibration when visualizing the icub model in rviz was very bad. We ran the calibration again with proper command of  calibrateWithType NPoseWalk` and the resulting calibration was good when checked through rviz. 

Ideally, we should receive an error when using a wrong command for calibration and we should double check how the calibration calls are being made and fix this issue.

@lrapetti @kouroshD @diegoferigo @claudia-lat @DanielePucci

Yeshasvitvs, 2019-11-18:
I double checked xsens wearable device [configuration file](https://github.com/robotology/wearables/blob/97fa49d5df9359c7cbc21f79928c3e5886af3326/app/xml/XsensSuitWearableDevice.xml#L18) and xsens manual for the calibration options. 

The correct format is `Npose` or  `NposeWalk`. So, the problem may not be what is stated in the comment above https://github.com/robotology/wearables/issues/66#issue-524276602 :confused:

We should double check the calibration a couple of times and see what is the problem.

diegoferigo, 2019-11-18:
In any case, if it's not already there, having an error to report that the RPC command is wrong is advisable.

lrapetti, 2019-11-18:
The point in which the calibration type string is passed is [here](https://github.com/robotology/wearables/blob/4ecd90c96e15708f3827467e833728ca23fa932c/XSensMVN/src/XSensMVNCalibrator.cpp#L181). We should check what is the value of the string at this point.

I checked also the documentation of `initializeCalibration` xsens call, and there is no information regarding what happen in case of wrong/non-existing type

[image]

we should check how we can check it