robotology/wearables #69: Xsens Wires Suit Does Not Calibrate

kouroshD, 2019-12-04:
Yesterday and today we tried a lot of times with different options to calibrate the Xsens suit, but it returns `poor` quality calibration results.  The result with `NposeWalk`, `Tpose` was always poor. The result with `Npose` was some times good, but the visualization result shows the poor result (as following you can see).
We have checked several times the sensor placements and they were correct.
[image]

kouroshD, 2019-12-04:
The result with frank wearing the Xsens wired suit. The result of the calibration in `Npose` was acceptable with frank.

[image]

lrapetti, 2019-12-05:
Given that previously we never experienced this, I'm not sure if the problem is related to software or to a failure in the hardware (broken sensors? Wrong connections?)

From the two gif you posted we don't have much information, since the behaviour is somehow normal when the calibration is bad. The only thing that is a bit weird is the rotation of the feet that is quite fast compared to the drift we observed in the past.

Yeshasvitvs, 2019-12-06:
One thing we can do is to try to run the wired suit with either @claudia-lat or @lrapetti and see if the calibration works without problems.

Yeshasvitvs, 2019-12-07:
Another viable solution is to request members who have xsens MVN studio license and try to calibrate the wired suit with it. Hopefully this will provide us more information on the status of the suit e.g., if there is any problem with the hardware.

DanielePucci, 2019-12-17:
@claudia-lat this is a very important issue we are currently having. Please discuss with @Yeshasvitvs @kouroshD @lrapetti and then come up with an action plan to follow and implement, thanks

claudia-lat, 2019-12-19:
[Updating 19/12/19]
After an internal discussion we decided to go for a test with the wired suits with other subjects (e.g., @claudia-lat @lrapetti).  This because the problem has been tested only with @kouroshD since now.

DanielePucci, 2020-01-02:
@claudia-lat any outcome of these tests?

claudia-lat, 2020-01-02:
> @claudia-lat any outcome of these tests?

Actually, the last available day in the lab before Christmas we didn't do the test. We can do the test the first day available with @lrapetti and @Yeshasvitvs

Yeshasvitvs, 2020-02-04:
Today @lrapetti and I tested the wired suit with @isorrentino as the subject. We tried both Npose and NposeWalk calibration and it went will without any problems.

[image]

CC @DanielePucci @claudia-lat 

@kouroshD can we close this issue now?

claudia-lat, 2020-02-11:
We can close the issue.

closed 2020-02-11 by claudia-lat

DanielePucci, 2020-02-24:
The fact that it restarted working without any specific fix is a kind of terrifying, bit we accept it

Yeshasvitvs, 2020-06-29:
> The fact that it restarted working without any specific fix is a kind of terrifying

We are experiencing the same problem again

CC @claudia-lat @lrapetti @kouroshD @DanielePucci