robotology/wearables #73: Prepare initial release

lrapetti, 2020-01-27:
Given that the code contained in `Wearable` can be considered being stable for a while, and given the discussion in https://github.com/robotology/robotology-superbuild/issues/331, I think we might do an initial release (v.1.0??) with the current version of the code.

@Yeshasvitvs @diegoferigo @traversaro what do you think?

traversaro, 2020-01-27:
I agree.

diegoferigo, 2020-01-27:
I would suggest following [semantic versioning](https://semver.org/) as much as possible. Since Wearable uses quite extensive interfaces, if we have pending modifications of the headers we should do them before the release.

We are using the project already for more than one year, and right now it's fairly stable. I would agree to release the first stable version `1.0` and freeze the APIs.

lrapetti, 2020-01-28:
Initial Release can be found at https://github.com/robotology/wearables/releases/tag/v1.0.0

closed 2020-01-28 by lrapetti