robotology/wearables #78: Compilation failed on iCubGenova04 laptop (icub30)

gabrielenava, 2020-02-03:
**Describe the bug**

The repo fails at compiling on the laptop associated to iCubGenova04 (10.0.0.30).

**To Reproduce**
Steps to reproduce the behavior:
1. Run `make wearables` in the robotology superbuild/build directory. Here is the error:

```
[ 90%] Performing build step for 'wearables'
[  3%] Built target SensorsImpl
[  9%] Built target XsensSuitControl
[ 71%] Built target WearableData
[ 73%] Building CXX object devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o
/usr/local/src/robot/robotology-superbuild/robotology/wearables/devices/IWearRemapper/src/IWearRemapper.cpp:12:10: fatal error: thrift/WearableData.h: No such file or directory
 #include "thrift/WearableData.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/build.make:86: recipe for target 'devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o' failed
make[6]: *** [devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/src/IWearRemapper.cpp.o] Error 1
CMakeFiles/Makefile2:334: recipe for target 'devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/all' failed
make[5]: *** [devices/IWearRemapper/CMakeFiles/IWearRemapper.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
CMakeFiles/wearables.dir/build.make:108: recipe for target 'robotology/wearables/CMakeFiles/YCMStamp/wearables-build' failed
make[3]: *** [robotology/wearables/CMakeFiles/YCMStamp/wearables-build] Error 2
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/wearables.dir/all' failed
make[2]: *** [CMakeFiles/wearables.dir/all] Error 2
CMakeFiles/Makefile2:190: recipe for target 'CMakeFiles/wearables.dir/rule' failed
make[1]: *** [CMakeFiles/wearables.dir/rule] Error 2
Makefile:214: recipe for target 'wearables' failed
make: *** [wearables] Error 2
``` 

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04 LTS

traversaro, 2020-02-03:
Which version of YARP and wearables are you using? This seems to be similar to https://github.com/robotology/wearables/issues/56 .

lrapetti, 2020-02-04:
> Which version of YARP and wearables are you using? This seems to be similar to #56 .

also which version of `wearables`

gabrielenava, 2020-02-04:
`wearables` is on `master` branch up to date with the last commit, while `YARP` is in a not-default branch: `feature/impl-static-friction-comp`. This may be the source of the issue. `git log` command shows me that the last commit in this branch is from October 30, 2019. I don't know if I can revert the branch of if this procedure requires further modifications.

lrapetti, 2020-02-04:
> `YARP` is in a not-default branch: `feature/impl-static-friction-comp`.

I think this is why `wearables` is failing then, as @traversaro was pointing a similar error was addressed as new YARP version were released. I don't know if we have to stay on that branch of `YARP`, probably @nunoguedelha can tell us.

claudia-lat, 2020-02-11:
@nunoguedelha do you have any suggestion on the previous comment?

lrapetti, 2020-02-17:
No feedback received, anyway all the CI checks are working fine so there should be no compilation problem in `wearable` itself. The problem raised, as mentioned, was probably due to an old/under development version of `YARP`.

If the problem persist with stable release of `YARP`, feel free to reopen the issue.

lrapetti, 2020-02-17:
Still @nunoguedelha, if you can, let us know if we can change the branch of `iCubGenova04` laptop so that we can use HDE on that PC

closed 2020-02-17 by lrapetti

traversaro, 2020-02-17:
I am not @nunoguedelha, but I guess that if the modifications in `feature/impl-static-friction-comp` are actually required, you can rebase it on the top of the latest `yarp-3.3` or `master` modifications of YARP, and use it to correctly compile the `wearables` repo.