robotology/wearables #79: Cleanup CMake code to install YARP devices

traversaro, 2020-05-24:
While testing the Windows installer for robotology-superbuild software, I executed the `yarpdev --list`  command and I noticed that wearables and human-dynamics installed device are not found correctly, while all the other YARP devices are correctly installed: 

<details>
 <summary>Click to see the outpiut of <code>yarpdev --list</code></summary>

~~~
STraversaro@IITICUBLAP102 MINGW64 ~
$ yarpdev --list
[INFO]Here are devices listed for your system:
error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWeard.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWeard.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWeard.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWeard.dll:
  The specified procedure could not be found.

[WARNING]Wrong library name for plugin ianalogsensor_to_iwear
error while opening C:/robotology/robotology/lib/yarp/ICub.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICubd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICub.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICubd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICub.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICubd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICub.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/ICubd.dll:
  The specified procedure could not be found.

[WARNING]Wrong library name for plugin icub_wearable_device
error while opening C:/robotology/robotology/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearRemapperd.dll:
  The specified procedure could not be found.

[WARNING]Wrong library name for plugin iwear_remapper
error while opening C:/robotology/robotology/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IWearWrapperd.dll:
  The specified procedure could not be found.

[WARNING]Wrong library name for plugin iwear_wrapper
error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapperd.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapperd.dll:
  The specified procedure could not be found.

[WARNING]Wrong library name for plugin ixsensmvncontrol_wrapper
[INFO]Device "AudioPlayerWrapper", available on request (found in C:/robotology/robotology/lib/yarp/yarp_AudioPlayerWrapper.dll library), is a network wrapper.
[INFO]Device "AudioRecorderWrapper", available on request (found in C:/robotology/robotology/lib/yarp/yarp_AudioRecorderWrapper.dll library), is a network wrapper.
[INFO]Device "JoypadControlClient", available on request (found in C:/robotology/robotology/lib/yarp/yarp_JoypadControlClient.dll library), is a network wrapper.
[INFO]Device "JoypadControlServer", available on request (found in C:/robotology/robotology/lib/yarp/yarp_JoypadControlServer.dll library), is a network wrapper.
[INFO]Device "RGBDSensorClient", available on request (found in C:/robotology/robotology/lib/yarp/yarp_RGBDSensorClient.dll library), is a network wrapper.
[...]
~~~

</details>

I did not investigated in detail why this is happening, but I noticed that in this repo manually wrote `.ini` files for all the devices are still installed, while YARP has gained support to automatically generated them (reduce the possibility of errors) in version 3.2.0 : https://github.com/robotology/yarp/blob/2522239984ef3e6c26939fe740001ec08189a8cb/doc/release/v3_2_0.md#build-system .

For this reason, I suggest to first of all remove the manually written `.ini` files, and use instead the much more convenient and less error prone automatic generation provided by YARP, see https://github.com/robotology/icub-main/pull/586 for an example pull request.

While we touch the CMake, I suggest also to call the library name of your devices in a less ambiguous way. For example, the `icub_wearable_device` YARP device installs a library called `ICub` in `<prefix>/lib/yarp/ICub.<ext>`, and as you imagine such a generic name is confusing for users and prone to collision with other libraries name, so called also the library  `<prefix>/lib/yarp/ICubWearable.<ext>` or something similar would be nice.

lrapetti, 2020-05-28:
> For this reason, I suggest to first of all remove the manually written `.ini` files, and use instead the much more convenient and less error prone automatic generation provided by YARP, see [robotology/icub-main#586](https://github.com/robotology/icub-main/pull/586) for an example pull request.

Done with https://github.com/robotology/wearables/pull/80, did it fixed the problem with `yarpdev --list` on Windows?

traversaro, 2020-05-28:
> Done with #80, did it fixed the problem with `yarpdev --list` on Windows?

I have no idea. : ) 
Next time I will regenerate and install the installer I will try it.

claudia-lat, 2020-06-01:
@lrapetti drop a comment here before closing.

lrapetti, 2020-06-03:
Closing via https://github.com/robotology/wearables/pull/80

closed 2020-06-03 by lrapetti

traversaro, 2020-11-17:
> I have no idea. : )
> Next time I will regenerate and install the installer I will try it.

To clarify, new version of wearables are included in the Windows binary only when they are released and the corresponding version is bumped in the .yaml file: https://github.com/robotology/robotology-superbuild/blob/master/releases/2020.11.yaml#L109 . At the moment, the v1.0.0 has been used for all the releases of the robotology-superbuild until now.