robotology/wearables #83: Implement outputPortName in IWearRemapper device

Yeshasvitvs, 2020-10-14:
During some tests with @CarlottaSartore we discovered that the [IWearRempper](https://github.com/robotology/wearables/tree/master/devices/IWearRemapper) device does not have any implementation in place for handling the [outputPortName](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/Human.xml#L17) configuration parameter that we use in different configuration files.

CC @lrapetti @claudia-lat

lrapetti, 2020-10-15:
I just realised that the output port of the `IWearRemapper` was removed with https://github.com/robotology/wearables/pull/34 following what stated by @diegoferigo in https://github.com/robotology/wearables/issues/27#issuecomment-467059753. 

The main reason was to have a clear distinction between:
- a `Remapper` device that read one/multiple `yarp ports` exposing an `interface`
- a `Wrapper` device that write a `yarp port` from an `interface`

We then should just update all the configuration files that still have `outputPort` configuration.

Yeshasvitvs, 2020-10-19:
A new issue is open at human-dynamics-estimation repo https://github.com/robotology/human-dynamics-estimation/issues/215
Closing this.

closed 2020-10-19 by Yeshasvitvs