robotology/wearables #90: Unable to compile when XSENS_MVN_USE_SDK is ON

GiulioRomualdi, 2020-11-17:
When I try to compile the wearables with the option `XSENS_MVN_USE_SDK` the compilation fails with a linking error.

```
1>------ Build started: Project: ZERO_CHECK, Configuration: Release x64 ------
2>------ Build started: Project: SensorsImpl, Configuration: Release x64 ------
3>------ Build started: Project: WearableData, Configuration: Release x64 ------
4>------ Build started: Project: XSensMVN, Configuration: Release x64 ------
5>------ Build started: Project: XsensSuitControl, Configuration: Release x64 ------
6>------ Build started: Project: IAnalogSensorToIWear, Configuration: Release x64 ------
7>------ Build started: Project: ICub, Configuration: Release x64 ------
3>Generating code from thrift/WearableData.thrift
2>SensorsImpl.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\Release\SensorsImpl.lib
7>ICub.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\yarp\Release\ICub.dll
4>XSensMVNDriverImpl.cpp
3>WearableData.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\Release\WearableData.lib
8>------ Build started: Project: IWearRemapper, Configuration: Release x64 ------
9>------ Build started: Project: IWearWrapper, Configuration: Release x64 ------
5>Generating code from thrift/XsensSuitControlService.thrift
8>IWearRemapper.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\yarp\Release\IWearRemapper.dll
9>IWearWrapper.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\yarp\Release\IWearWrapper.dll
6>IAnalogSensorToIWear.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\yarp\Release\IAnalogSensorToIWear.dll
5>XsensSuitControl.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\Release\XsensSuitControl.lib
10>------ Build started: Project: IXsensMVNControlWrapper, Configuration: Release x64 ------
4>C:\Program Files\Xsens\MVN SDK 2018.0.3\SDK Files\x64\include\xsens/xsstring.h(329,32): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
4>C:\Program Files\Xsens\MVN SDK 2018.0.3\SDK Files\x64\include\xsens/xsexception.h(49,4): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
10>IXsensMVNControlWrapper.vcxproj -> C:\robotology-superbuild\build\robotology\wearables\lib\yarp\Release\IXsensMVNControlWrapper.dll
4>C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.26.28801\include\experimental/filesystem(30,1): fatal error C1189: #error:  The <experimental/filesystem> header providing std::experimental::filesystem is deprecated by Microsoft and will be REMOVED. It is superseded by the C++17 <filesystem> header providing std::filesystem. You can define _SILENCE_EXPERIMENTAL_FILESYSTEM_DEPRECATION_WARNING to acknowledge that you have received this warning.
4>Done building project "XSensMVN.vcxproj" -- FAILED.
11>------ Build started: Project: XsensSuit, Configuration: Release x64 ------
12>------ Build started: Project: testCalibrator, Configuration: Release x64 ------
13>------ Build started: Project: testDriver, Configuration: Release x64 ------
12>Building Custom Rule C:/robotology-superbuild/robotology/wearables/XSensMVN/test/CMakeLists.txt
11>Building Custom Rule C:/robotology-superbuild/robotology/wearables/devices/XsensSuit/CMakeLists.txt
11>XsensSuit.cpp
12>testCalibrator.cpp
13>Building Custom Rule C:/robotology-superbuild/robotology/wearables/XSensMVN/test/CMakeLists.txt
13>testDriver.cpp
12>C:\Program Files\Xsens\MVN SDK 2018.0.3\SDK Files\x64\include\xsens/xsstring.h(329,32): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
11>C:\robotology-superbuild\robotology\wearables\interfaces\IWear\include\Wearable/IWear/IWear.h(61,5): warning C5208: unnamed class used in typedef name cannot declare members other than non-static data members, member enumerations, or member classes
12>C:\Program Files\Xsens\MVN SDK 2018.0.3\SDK Files\x64\include\xsens/xsexception.h(49,4): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
11>C:\robotology-superbuild\build\install\include\yarp/dev/PreciselyTimed.h(14,1): warning C4083: expected 'string'; found identifier '__LOC__'
13>C:\Program Files\Xsens\MVN SDK 2018.0.3\SDK Files\x64\include\xsens\xsstring.h(329,32): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
13>C:\Program Files\Xsens\MVN SDK 2018.0.3\SDK Files\x64\include\xsens/xsexception.h(49,4): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
12>LINK : fatal error LNK1181: cannot open input file '..\..\lib\Release\XSensMVN.lib'
12>Done building project "testCalibrator.vcxproj" -- FAILED.
13>LINK : fatal error LNK1181: cannot open input file '..\..\lib\Release\XSensMVN.lib'
13>Done building project "testDriver.vcxproj" -- FAILED.
11>LINK : fatal error LNK1181: cannot open input file '..\..\lib\Release\XSensMVN.lib'
11>Done building project "XsensSuit.vcxproj" -- FAILED.
14>------ Build started: Project: ALL_BUILD, Configuration: Release x64 ------
14>Building Custom Rule C:/robotology-superbuild/robotology/wearables/CMakeLists.txt
========== Build: 10 succeeded, 4 failed, 0 up-to-date, 0 skipped ==========
```

The xsens sdk has been installed following https://github.com/robotology/human-dynamics-estimation/wiki/Set-up-Machine-for-running-HDE#xsens-only-for-windows 

Here the cmake configuration related to xsens
[image]

**Desktop (please complete the following information):**
 - OS: Windows10

cc @kouroshD @Yeshasvitvs @lrapetti @traversaro @s-dafarra

S-Dafarra, 2020-11-17:
> 13>LINK : fatal error LNK1181: cannot open input file '..\..\lib\Release\XSensMVN.lib'

Have you checked if that file exists?

GiulioRomualdi, 2020-11-17:
I don't know if it's related but if I compiled wearables from the superbuild I get
[image]

traversaro, 2020-11-17:
> 
> 
> I don't know if it's related but if I compiled wearables from the superbuild I get
> [image]

I guess we can just switch to use C++17 and fix the error, right?

GiulioRomualdi, 2020-11-17:
@S-Dafarra updated ycm to ycm-0.12 and then he recompiled the superbuild. After that, the error disappeared.

traversaro, 2020-11-17:
> 
> 
> @S-Dafarra updated ycm to ycm-0.12 and then he recompiled the superbuild. After that, the error disappeared.
> 
> I don't know if this is related

This does not seem to be related. Are you able to compile wearables on its own and also the XSensMVN library even by fixing the experimental/filesystem usage?

traversaro, 2020-11-17:
Actually the error should have been suppressed by https://github.com/robotology/wearables/pull/82, are you sure that you simply did not have an older commit of wearables, or that you not did run CMake again after updating wearables?

lrapetti, 2020-11-24:
> Actually the error should have been suppressed by #82, are you sure that you simply did not have an older commit of wearables, or that you not did run CMake again after updating wearables?

@GiulioRomualdi do you think this can be possibly the cause of your problem?
Can we close this issue?

claudia-lat, 2020-12-01:
CC @GiulioRomualdi

Yeshasvitvs, 2020-12-04:
After a recent git pull from upstream, the compilation was good but a new issue https://github.com/robotology/wearables/issues/91 was observed by @GiulioRomualdi 

So, this issue can be closed. (Checked with @GiulioRomualdi)

closed 2020-12-04 by Yeshasvitvs