robotology/wearables #91: Unable to load the yarp devices built by wearables

GiulioRomualdi, 2020-11-17:
We installed wearables with the superbuild however when we try to check the devices `yarp dev --list` the following errors occur 
```
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IAnalogSensorToIWear.dll:
  The specified procedure could not be found.

[WARNING] |yarp.dev.Drivers| Wrong library name for plugin ianalogsensor_to_iwear
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/ICub.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/ICub.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/ICub.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/ICub.dll:
  The specified procedure could not be found.

[WARNING] |yarp.dev.Drivers| Wrong library name for plugin icub_wearable_device
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearRemapper.dll:
  The specified procedure could not be found.

[WARNING] |yarp.dev.Drivers| Wrong library name for plugin iwear_remapper
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IWearWrapper.dll:
  The specified procedure could not be found.

[WARNING] |yarp.dev.Drivers| Wrong library name for plugin iwear_wrapper
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/IXsensMVNControlWrapper.dll:
  The specified procedure could not be found.

[WARNING] |yarp.dev.Drivers| Wrong library name for plugin ixsensmvncontrol_wrapper
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/XsensSuit.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/XsensSuit.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/XsensSuit.dll:
  The specified procedure could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology-superbuild/build/install/lib/yarp/XsensSuit.dll:
  The specified procedure could not be found.
```

I'm using windows 10

traversaro, 2020-11-17:
Which version of YARP are you using?

traversaro, 2020-11-17:
The error:

> Wrong library name for plugin ixsensmvncontrol_wrapper

Was added to solve the issue https://github.com/robotology/yarp/issues/975 . Can you double check the content of the automatically generated .ini files installed by wearables for YARP devices?

traversaro, 2020-11-17:
I see that this repo is still shipping manually written .ini files for YARP devices. While this in theory is still supported, is for sure much more error prone. Switching to the automatic generation of this file should avoid possible problems.

traversaro, 2020-11-17:
Related issue: https://github.com/robotology/wearables/issues/79 .

traversaro, 2020-11-17:
> 
> 
> I see that this repo is still shipping manually written .ini files for YARP devices. While this in theory is still supported, is for sure much more error prone. Switching to the automatic generation of this file should avoid possible problems.

In theory this was fixed in https://github.com/robotology/wearables/pull/80 . I am not sure why .ini files are still around.

traversaro, 2020-11-17:
> In theory this was fixed in #80 . I am not sure why .ini files are still around.

Sorry, I probably was wrong and instead the files are no longer there.

GiulioRomualdi, 2020-11-17:
> Which version of YARP are you using?

YARP version 3.4.1+29-20201020.2+git35f80d6d4

GiulioRomualdi, 2020-11-17:
> Sorry, I probably was wrong and instead the files are no longer there.

Yes, looks like the fix was done only to `devices` and not `wrappers`.

traversaro, 2020-11-17:
In any case, I think it would be useful to check all the files installed by wearables and check if there is anything strange there. I am particularly worried about the `ICub`  device, that being such an ambiguous name (as noted in https://github.com/robotology/wearables/issues/79) could create problems with similar named configuration files in case insensitive filesystems.

traversaro, 2020-11-17:
All the other devices installed by the superbuild instead listed by `yarp dev --list` with no problems?

DanielePucci, 2020-11-17:
@Yeshasvitvs @lrapetti may you give your inputs here?

Yeshasvitvs, 2020-11-17:
I am here with @GiulioRomualdi , it seems like the installation of the plugins of wearable repo is not done correctly.

Yeshasvitvs, 2020-11-17:
> All the other devices installed by the superbuild instead listed by yarp dev --list with no problems?

yes, only the plugins of the wearable library could not be run

traversaro, 2020-11-18:
Given that the demo was done successfully, I guess that this was solved, or at least a workaround was found?

Yeshasvitvs, 2020-11-19:
@GiulioRomualdi please close this if do not have the problem anymore.

traversaro, 2020-11-19:
To be honest I was able to reproduce the problem as part of the robotology-superbuild CI in https://github.com/robotology/robotology-superbuild/pull/521/checks?check_run_id=1425070305 (see " Check if YARP devices are correctly listed [Windows]" section, part of PR https://github.com/robotology/robotology-superbuild/pull/521):
~~~
2020-11-19T17:00:54.0647419Z [INFO] |yarp.dev.Drivers| Here are devices listed for your system:
2020-11-19T17:00:54.2423470Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWear.dll:
2020-11-19T17:00:54.2425179Z   The specified procedure could not be found.
2020-11-19T17:00:54.2425697Z 
2020-11-19T17:00:54.2426213Z 
2020-11-19T17:00:54.2429587Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IAnalogSensorToIWear.dll:
2020-11-19T17:00:54.2431515Z   The specified procedure could not be found.
2020-11-19T17:00:54.2432054Z 
2020-11-19T17:00:54.2432380Z 
2020-11-19T17:00:54.2472351Z [WARNING] |yarp.dev.Drivers| Wrong library name for plugin ianalogsensor_to_iwear
2020-11-19T17:00:54.2484665Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/ICub.dll:
2020-11-19T17:00:54.2485871Z   The specified procedure could not be found.
2020-11-19T17:00:54.2486390Z 
2020-11-19T17:00:54.2486734Z 
2020-11-19T17:00:54.2491262Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/ICub.dll:
2020-11-19T17:00:54.2492460Z   The specified procedure could not be found.
2020-11-19T17:00:54.2492958Z 
2020-11-19T17:00:54.2493315Z 
2020-11-19T17:00:54.2531497Z [WARNING] |yarp.dev.Drivers| Wrong library name for plugin icub_wearable_device
2020-11-19T17:00:54.2560709Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IWearRemapper.dll:
2020-11-19T17:00:54.2565023Z   The specified procedure could not be found.
2020-11-19T17:00:54.2565566Z 
2020-11-19T17:00:54.2565971Z 
2020-11-19T17:00:54.2567066Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IWearRemapper.dll:
2020-11-19T17:00:54.2568207Z   The specified procedure could not be found.
2020-11-19T17:00:54.2568703Z 
2020-11-19T17:00:54.2569026Z 
2020-11-19T17:00:54.2608016Z [WARNING] |yarp.dev.Drivers| Wrong library name for plugin iwear_remapper
2020-11-19T17:00:54.2621222Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IWearWrapper.dll:
2020-11-19T17:00:54.2622494Z   The specified procedure could not be found.
2020-11-19T17:00:54.2623312Z 
2020-11-19T17:00:54.2623659Z 
2020-11-19T17:00:54.2627984Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IWearWrapper.dll:
2020-11-19T17:00:54.2629187Z   The specified procedure could not be found.
2020-11-19T17:00:54.2629693Z 
2020-11-19T17:00:54.2631649Z 
2020-11-19T17:00:54.2675907Z [WARNING] |yarp.dev.Drivers| Wrong library name for plugin iwear_wrapper
2020-11-19T17:00:54.2689554Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapper.dll:
2020-11-19T17:00:54.2691374Z   The specified procedure could not be found.
2020-11-19T17:00:54.2691893Z 
2020-11-19T17:00:54.2692202Z 
2020-11-19T17:00:54.2697888Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/yarp/IXsensMVNControlWrapper.dll:
2020-11-19T17:00:54.2699345Z   The specified procedure could not be found.
2020-11-19T17:00:54.2699950Z 
2020-11-19T17:00:54.2700310Z 
2020-11-19T17:00:54.2740612Z [WARNING] |yarp.dev.Drivers| Wrong library name for plugin ixsensmvncontrol_wrapper
2020-11-19T17:00:54.3549673Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/iCub/esdCan.dll:
2020-11-19T17:00:54.3551267Z   The specified module could not be found.
2020-11-19T17:00:54.3551959Z 
2020-11-19T17:00:54.3552366Z 
2020-11-19T17:00:54.3587475Z [WARNING] |yarp.dev.Drivers| Wrong library name for plugin ecan
2020-11-19T17:00:54.3902672Z [INFO] |yarp.dev.Drivers| Device "AudioPlayerWrapper", available on request (found in C:/robotology/robotology/lib/yarp/yarp_AudioPlayerWrapper.dll library), is a network wrapper.
2020-11-19T17:00:54.3904860Z [INFO] |yarp.dev.Drivers| Device "AudioRecorderWrapper", available on request (found in C:/robotology/robotology/lib/yarp/yarp_AudioRecorderWrapper.dll library), is a network wrapper.
2020-11-19T17:00:54.3907164Z [INFO] |yarp.dev.Drivers| Device "IMURosPublisher", available on request (found in C:/robotology/robotology/lib/yarp/yarp_IMURosPublisher.dll library), is a network wrapper.
2020-11-19T17:00:54.3909097Z [INFO] |yarp.dev.Drivers| Device "JoypadControlClient", available on request (found in C:/robotology/robotology/lib/yarp/yarp_JoypadControlClient.dll library), is a network wrapper.
2020-11-19T17:00:54.3911350Z [INFO] |yarp.dev.Drivers| Device "JoypadControlServer", available on request (found in C:/robotology/robotology/lib/yarp/yarp_JoypadControlServer.dll library), is a network wrapper.
2020-11-19T17:00:54.3913518Z [INFO] |yarp.dev.Drivers| Device "MagneticFieldRosPublisher", available on request (found in C:/robotology/robotology/lib/yarp/yarp_MagneticFieldRosPublisher.dll library), is a network wrapper.
[... cut ...]
2020-11-19T17:00:54.4086994Z [INFO] |yarp.dev.Drivers| Device "parametricCalibratorEth", available on request (found in C:/robotology/robotology/lib/iCub/parametricCalibratorEth.dll library).
2020-11-19T17:00:54.4088810Z [INFO] |yarp.dev.Drivers| Device "sharedcan", available on request (found in C:/robotology/robotology/lib/iCub/shcan.dll library).
2020-11-19T17:00:54.4090211Z [INFO] |yarp.dev.Drivers| Device "skinWrapper", available on request (found in C:/robotology/robotology/lib/iCub/skinWrapper.dll library).
~~~

DanielePucci, 2020-11-20:
@Yeshasvitvs @lrapetti @GiulioRomualdi, it would be nice that someone gives an update on this 
> Given that the demo was done successfully, I guess that this was solved, or at least a workaround was found?

to avoid future problems in hard times

prashanthr05, 2020-11-20:
> Given that the demo was done successfully, I guess that this was solved, or at least a workaround was found?

Sorry, we got carried away with the results of the experiments. No, this issue was not solved.  We need to work on this and understand the problem.

As a first step, maybe we can try to check the versions of the software on the other Windows machine on which we ran the experiment and try to replicate that environment in the newer machine (which was supposed to be used for the experiments.)

prashanthr05, 2020-11-20:
> @GiulioRomualdi please close this if do not have the problem anymore.

I think we will forget about this issue if we close it, let's keep it open until we solve it.

GiulioRomualdi, 2020-11-20:
The problem is not solved. We ran the experiments on a different laptop.

I would avoid to close this issue

traversaro, 2020-11-21:
> We ran the experiments on a different laptop.

Just to understand, given that the problem was present was already in May https://github.com/robotology/wearables/issues/79, that laptop was not updated since before May? Which version of YARP, CMake and wearables is that laptop using?

lrapetti, 2020-11-23:
> > We ran the experiments on a different laptop.
> 
> Just to understand, given that the problem was present was already in May #79, that laptop was not updated since before May? Which version of YARP, CMake and wearables is that laptop using?

I am not sure about the setup we are talking about above, but in the new setup which was prepared from scratch in Sept/Oct by @Yeshasvitvs, the device are loaded correctly. We should investigate why there is working fine.

traversaro, 2020-11-23:
> I am not sure about the setup we are talking about above, but in the new setup which was prepared from scratch in Sept/Oct by @Yeshasvitvs, the device are loaded correctly. We should investigate why [...]

That is quite interesting. A few info that it would be interesting to get there: 
* Version of the dependency and how have been obtained 
* Version of CMake 
* Version of YCM
* Version of YARP

Yeshasvitvs, 2020-11-23:
> That is quite interesting. A few info that it would be interesting to get there: 

> * Version of the system dependencies such as ACE and how have been obtained (old installers, vcpkg, vcpkg binary archive) 

I double checked the dependencies, and I used https://github.com/microsoft/vcpkg/ at https://github.com/microsoft/vcpkg/commit/5d33bd3d7910c6090911f71f9dcf1a4eecd303da

May be this needs to be aligned with the documentation on robotology https://github.com/robotology/robotology-superbuild#system-libraries and use  https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/releases
 
> * Version of CMake

CMake 3.18.3
 
> * Version of YCM

https://github.com/robotology/ycm/tree/ycm-0.11

> * Version of YARP

https://github.com/robotology/yarp/tree/yarp-3.4

Another detail as pointed by @traversaro  is the last release tag of wearables https://github.com/robotology/wearables/releases/tag/v1.0.0 that is used in https://github.com/robotology/robotology-superbuild/blob/master/releases/2020.11.yaml#L106-L109

We are working on a bit of cleanup now following which @lrapetti with make a new release and test the new release on the machine @GiulioRomualdi ran into problems.

traversaro, 2020-11-23:
> UPDATE: I was doing some development #82 #84 #81 on the machine and it was updated to master branch latest commit e7429bc

And it still works fine, i.e. `yarpdev --list` does not have the problem described in the issue?

Yeshasvitvs, 2020-11-23:
> And it still works fine, i.e. yarpdev --list does not have the problem described in the issue?

yes, all the plugins (devices and wrappers) that are built and installed from wearables repo are discovered correctly.

claudia-lat, 2020-12-01:
The idea can be to investigate if the errore is reproducible on a new installation on Windows. 
@GiulioRomualdi if you agree we can move this issue on Backlog. Let us know.

traversaro, 2020-12-02:
> I am not sure about the setup we are talking about above, but in the new setup which was prepared from scratch in Sept/Oct by @Yeshasvitvs, the device are loaded correctly. We should investigate why there is working fine.

Can you please refer to machines with their IIT code so that it is clear to which machine are we referring to? 

> > And it still works fine, i.e. yarpdev --list does not have the problem described in the issue?
> 
> yes, all the plugins (devices and wrappers) that are built and installed from wearables repo are discovered correctly.

Was that on a clean build or not? 
Can you please specify the exact checkout of YARP/YCM/Wearables/Windows that you used? In the previous comment you just linked the **branches** of YARP/YCM, but those do not specify the exact version, you should specify the exact commit. 

To clarify, the fact that on **some machines** it is working, and that on other machines and on GitHub Actions (that is a perfectly deterministic environment) it is not working, lead me to think that there is probably either something that depends on the environment (such as some local CMake or environment change) of the specific machine, or (even worse!) a non-deterministic issue. If it was the other way around (i.e. that on GitHub actions it did work, and just on some machines it was not working) I would be more relaxed, but the fact that the problem is **constantly occuring** on GitHub Actions lead me to think that it is highly probably that it could occur on new Windows machine, and I think that having software that "sometimes" does not work is a ticking bomb that could create problems in the future (think for example if we need to prepare a new laptop for Andy/iFeel demos).

claudia-lat, 2020-12-02:
@Yeshasvitvs @lrapetti @kouroshD may we have a quick call today to understand the problem? Maybe it was not clear to me yesterday and it's better to clarify it with you.

GiulioRomualdi, 2020-12-02:
Hi, guys at this moment. I don't have time to check this. I'm sorry. If you don't want to keep the issue open please feel free to close it. Someone will reopen it again in case of problems

claudia-lat, 2020-12-02:
No worries @GiulioRomualdi .  It's better for us (An.Dy guys) to have a check on this. Thanks anyway.

lrapetti, 2020-12-02:
> > I am not sure about the setup we are talking about above, but in the new setup which was prepared from scratch in Sept/Oct by @Yeshasvitvs, the device are loaded correctly. We should investigate why there is working fine.
> 
> Can you please refer to machines with their IIT code so that it is clear to which machine are we referring to?
> ...
> Was that on a clean build or not?

We are referring to `IITICUBLAP177`, which we got new so it was a clean laptop with clean build

 > Can you please specify the exact checkout of YARP/YCM/Wearables/Windows that you used?

Here are the exact commit:
- Windows Version `10.0.18362`
- YARP commit: https://github.com/robotology/yarp/commit/19683ac4e2145f404992dd4be57532b65cbd988c
- YCM commit: https://github.com/robotology/ycm/commit/a8916a4229e01796225d304785ecb4ae2782d1ab
- wearables commit: https://github.com/robotology/wearables/commit/e7429bc99833b49e26d8dbb7827cd9b7df66fe7f

We could try to replicate the same configuration to investigate if the problem was introduced by some update in YCM or YARP, or try to update them to check if the error occur also on this machine.

traversaro, 2020-12-02:
I restarted the build in https://github.com/robotology/robotology-superbuild/pull/521/checks?check_run_id=1485361566, and the problem is not occurring anymore in the `Stable` and `Unstable` branches, and is still occurring in the 2020.11 that was still using the v1.0.0 release (the 2020.11 build on master instead should use v1.1.0). So it is possible that the problem was solved by some of the latest PRs to wearables, but probably it is a good idea to have the same check to Wearable's CI to be sure. I will rebase https://github.com/robotology/robotology-superbuild/pull/521 on the top of the latest master commit to see if the problem is also solved with the latest 2020.11 releae as well.

lrapetti, 2020-12-02:
> So it is possible that the problem was solved by some of the latest PRs to wearables

Possibly replacing `.ini` file with autogenerated files (https://github.com/robotology/wearables/pull/80, andhttps://github.com/robotology/wearables/pull/93)

> but probably it is a good idea to have the same check to Wearable's CI to be sure.

I'll open a separate issue for adding this check https://github.com/robotology/wearables/issues/98

traversaro, 2020-12-02:
Perfect!

Yeshasvitvs, 2020-12-04:
> Possibly replacing .ini file with autogenerated files (#80, and #93)

Following these changes in the upstream, I compiled and installed wearables on the Alienware machine this issue was observed by @GiulioRomualdi . All the wearable devices are now discovered and loaded correctly.

[image]

[image]

**NOTE**: For the `xsens_suit` plugin to load while running `yarp plugin --all` the xsens key needs to be plugged in

We can do one more final testing of launching [xsens suit wearable device](https://github.com/robotology/wearables/blob/master/app/xml/XsensSuitWearableDevice.xml) on this machine before closing this issue, to ensure everything is in order.

CC @lrapetti @S-Dafarra @GiulioRomualdi @claudia-lat @traversaro @DanielePucci

Yeshasvitvs, 2020-12-04:
> We can do one more final testing of launching xsens suit wearable device on this machine before closing this issue, to ensure everything is in order.

I did a quick test with the xsens sensors placed on the table, and the xsens suit wearable device loaded, calibrated and started data acquisition without any problems

[image]

traversaro, 2020-12-04:
Perfect, thanks @Yeshasvitvs ! At this point we can close the issue I guess?

Yeshasvitvs, 2020-12-04:
Closing after checking with @GiulioRomualdi

closed 2020-12-04 by Yeshasvitvs