robotology/wearables #100: Discuss wearables actuators feedback

lrapetti, 2020-12-23:
This library was initially intended to be used for handling **wearable sensors** data.

Recently we discussed the possibility to implement also the communication to be used for **wearable actuators**. This point should be discussed in order to define if/how to implement this feature along with the current wearable infrastructure.

lrapetti, 2020-12-23:
It would be good to collect some opinion and feedback on possible implementation solution from @Yeshasvitvs @diegoferigo @traversaro.

We are also planning to organize an _in person_ meeting to discuss this point.

traversaro, 2020-12-23:
I imagine this is related to some haptic device mounted on the wearable device, right? I think that having at least a small description of the kind of device that you want to control (haptic in the sense of a vibration? A vibration with different patterns? or an haptic device in which you can specify the intensity and/or the direction of the force?) can be useful to kickstart the discussion.

Yeshasvitvs, 2020-12-23:
@lrapetti @diegoferigo @traversaro I added a discussions tab to wearable library and would like to move this discussion there if it is not a problem https://github.com/robotology/wearables/discussions/103

> an haptic device in which you can specify the intensity and/or the direction of the force?) can be useful to kickstart the discussion.

@traversaro I added an example user story using the adaptive Paexo device we will be working on. Please check the discussion in https://github.com/robotology/wearables/discussions/103 and let us know if any further information is needed to make it clearer

Yeshasvitvs, 2021-01-25:
This issue is handled in https://github.com/robotology/wearables/issues/104
Closing

closed 2021-01-25 by Yeshasvitvs