robotology/wearables #127: Fix remapper plus logger termination

lrapetti, 2021-07-29:
We have observed that launching an application with the `IWearRemapper` plus the `IWearLogger`, we were not able to terminate it correctly. In particular, we observed that the logger was able to finish the `close()` and reach the end of the `destructor` (so it seems to be closing fine), but the remapper didn't close.

Probably the problem is not at all related to the logger and can happen even in different situation with the remapper, but this should be verified.

cc @kouroshD @prashanthr05 @Yeshasvitvs

Yeshasvitvs, 2021-07-29:
@lrapetti if you can, please add the termination log details. Thanks.

lrapetti, 2021-10-14:
I think this problem was fixed, do you remember something @prashanthr05 ?

prashanthr05, 2021-10-14:
@lrapetti I remember you telling me that you had verified that the termination problem was not related to logger. At that time, you were continuing to verify the remapper.

lrapetti, 2021-10-15:
> @lrapetti I remember you telling me that you had verified that the termination problem was not related to logger. At that time, you were continuing to verify the remapper.

Yes, actually now I remember that the logger was saving the data so it was basically reaching the termination phase. I will try to investigate again and add more information here.