robotology/wearables #128: Xsens suit device not starting

lrapetti, 2021-07-29:
**Describe the bug**
I have compield `wearables` trough the `robotology-superbuild` with the `XSENS_MVN_USE_SDK ` flag on. I have connected the station and the license dongle, but the device is terminating during the configuration without giving any error message:

```
PS C:\Users\icub\dev\robotology-superbuild\src\wearables\app\xml> yarprobotinterface.exe --config .\XsensSuitWearableDev
ice.xml
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[DEBUG] Reading file C:\Users\icub\dev\robotology-superbuild\src\wearables\app\xml\.\XsensSuitWearableDevice.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file C:\Users\icub\dev\robotology-superbuild\src\wearables\app\xml\.\XsensSuitWearableDevice.xml
[DEBUG] Preprocessor complete in:  1.459e-06 s
[INFO] |yarp.os.Port| Port //yarprobotinterface active at tcp://10.240.9.112:10002/
[INFO] startup phase starting...
[INFO]   Starting to configure
PS C:\Users\icub\dev\robotology-superbuild\src\wearables\app\xml>
```

lrapetti, 2021-07-29:
I did a first investigation, and it seems that the application exit [here](https://github.com/robotology/wearables/blob/master/devices/XsensSuit/src/XsensSuit.cpp#L650) when trying to check for configuration options. The problem seems to happen even if we check for other options (e.g. skipping to [the next check](https://github.com/robotology/wearables/blob/master/devices/XsensSuit/src/XsensSuit.cpp#L657))

kouroshD, 2021-08-02:
@lrapetti  The problem was that, they were built with `debug`option. So, I fixed them with `Release option` and they are working now.
I could run the xsens wearable device as in the picture, and do a dummy calibration and stream the data.

[image]
 
Laptop: Old AnDy laptop: IITICUBLAP076

closed 2021-10-14 by claudia-lat