robotology/wearables #136: problem with `IWear::getVirtualLinkKinSensor(sensorName)`

kouroshD, 2021-11-02:
**Describe the bug**
If I pass a wrong name to the interface method to get the pointer to the sensor associated with the passed name, I would expect to return a valid sensor pointer iff the sensorName passed is valid and there exist a sensor with that name. Otherwise, I would expect an error, nullptr, or at least a warning.

lrapetti, 2022-02-02:
Hi @kouroshD, is this bug still there?

Since [`getVirtualLinkKinSensor()`](https://github.com/robotology/wearables/blob/efb34b1baaf98006ef2e9add0cdfe98ec6f30e86/interfaces/IWear/include/Wearable/IWear/IWear.h#L183) in general is a pure virtual method, the problem should be related to the specific device you were running. In case can you specify what did you try to run?

lrapetti, 2022-10-19:
As far as I see in `IWearRemapper` the method [`getVirtualLinkKinSensor `](https://github.com/robotology/wearables/blob/3804cf7786e94f137fb9893763121a2de8dd3fb4/devices/IWearRemapper/src/IWearRemapper.cpp#L1073-L1080) is calling [`getSensor `](https://github.com/robotology/wearables/blob/3804cf7786e94f137fb9893763121a2de8dd3fb4/devices/IWearRemapper/src/IWearRemapper.cpp#L716-L725) which returns a `nullptr` if the sensor is not present.

I don't know if this issue was still a problem, but I would close it and in case re-open in case any related issue is observed again.

cc @RiccardoGrieco