robotology/wearables #149: HapticGlove device fails to start

S-Dafarra, 2022-04-04:
**Describe the bug**

I have enabled ``HapticGlove`` on ``wearables``, but when I try to launch 
```
yarprobotinterface --config HapticGlove.xml
```
I get the following output
```
(robsub) D:\Windows>yarprobotinterface --config HapticGlove.xml
[DEBUG] Reading file D:\Windows\robotology-superbuild\build\install\share\Wearables\HapticGlove.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file D:\Windows\robotology-superbuild\build\install\share\Wearables\HapticGlove.xml
[DEBUG] Preprocessor complete in:  7.5e-06 s
[INFO] |yarp.os.Port|//yarprobotinterface| Port //yarprobotinterface active at tcp://10.0.0.127:10012/
[INFO] startup phase starting...
[INFO] Opening device LeftHapticGloveWearableDevice with parameters [("robotName" = ""), ("period" = "0.01"), ("rightHand" = "false"), ("human_joint_list" = "( "l_thumb_oppose", "l_thumb_proximal", "l_thumb_middle", "l_thumb_distal", "l_index_abduction", "l_index_proximal", "l_index_middle", "l_index_distal", "l_middle_abduction", "l_middle_proximal", "l_middle_middle", "l_middle_distal", "l_ring_abduction", "l_ring_proximal", "l_ring_middle", "l_ring_distal", "l_pinky_abduction", "l_pinky_proximal", "l_pinky_middle", "l_pinky_distal" )"), ("human_finger_list" = "( "l_thumb_finger", "l_index_finger", "l_middle_finger", "l_ring_finger", "l_little_finger" )"), ("hand_link" = "l_hand")]
[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[WARNING] |yarp.os.YarpPluginSettings| Cannot load plugin from shared library (HapticGlove)
[WARNING] |yarp.os.YarpPluginSettings| (The specified module could not be found.

The specified module could not be found.

The specified module could not be found.
)
[ERROR] |yarp.dev.PolyDriver|LeftHapticGloveWearableDevice| Could not find device <hapticGlove>
[WARNING] Cannot open device LeftHapticGloveWearableDevice
[WARNING] Cannot open device LeftHapticGloveWearableDevice
[INFO] Opening device LeftHapticGloveWearableDeviceWrapper with parameters [("robotName" = ""), ("period" = "0.01"), ("dataPortName" = "/WearableData/HapticGlove/LeftHand/data:o"), ("rpcPortName" = "/WearableData/HapticGlove/LeftHand/metadataRpc:o")]
[DEBUG] |yarp.dev.PolyDriver|LeftHapticGloveWearableDeviceWrapper| Parameters are (dataPortName "/WearableData/HapticGlove/LeftHand/data:o") (device iwear_wrapper) (id LeftHapticGloveWearableDeviceWrapper) (period 0.01) (robotName) (rpcPortName "/WearableData/HapticGlove/LeftHand/metadataRpc:o")
[INFO] |yarp.dev.PolyDriver|LeftHapticGloveWearableDeviceWrapper| Created device <iwear_wrapper>. See C++ class wearable::wrappers::IWearWrapper for documentation.
[INFO] Opening device LeftHapticGloveWearableDeviceActuatorsWrapper with parameters [("robotName" = ""), ("period" = "0.01"), ("actuatorCommandInputPortName" = "/WearableData/HapticGlove/LeftHand/Actuators/input:i")]
[DEBUG] |yarp.dev.PolyDriver|LeftHapticGloveWearableDeviceActuatorsWrapper| Parameters are (actuatorCommandInputPortName "/WearableData/HapticGlove/LeftHand/Actuators/input:i") (device iwearactuators_wrapper) (id LeftHapticGloveWearableDeviceActuatorsWrapper) (period 0.01) (robotName)
[INFO] |yarp.os.Port|/WearableData/HapticGlove/LeftHand/Actuators/input:i| Port /WearableData/HapticGlove/LeftHand/Actuators/input:i active at tcp://10.0.0.127:10318/
[INFO] |yarp.dev.PolyDriver|LeftHapticGloveWearableDeviceActuatorsWrapper| Created device <iwearactuators_wrapper>. See C++ class wearable::wrappers::IWearActuatorsWrapper for documentation.
[INFO] Opening device RightHapticGloveWearableDevice with parameters [("robotName" = ""), ("period" = "0.01"), ("rightHand" = "true"), ("human_joint_list" = "( "r_thumb_oppose", "r_thumb_proximal", "r_thumb_middle", "r_thumb_distal", "r_index_abduction", "r_index_proximal", "r_index_middle", "r_index_distal", "r_middle_abduction", "r_middle_proximal", "r_middle_middle", "r_middle_distal", "r_ring_abduction", "r_ring_proximal", "r_ring_middle", "r_ring_distal", "r_pinky_abduction", "r_pinky_proximal", "r_pinky_middle", "r_pinky_distal" )"), ("human_finger_list" = "( "r_thumb_finger", "r_index_finger", "r_middle_finger", "r_ring_finger", "r_little_finger" )"), ("hand_link" = "r_hand")]
[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.

[... 81 lines cut ...]

**To Reproduce**
I have enabled ``HapticGlove`` and then compiled and installed. I am using the ``robotology-superbuild``.

**Expected behavior**
The error
```
 [ERROR] |yarp.os.YarpPluginSettings| Error while opening D:/Windows/robotology-superbuild/build/install/lib/yarp/HapticGlove.dll:
  The specified module could not be found.
```
should not occur. The library seems to be there 
[image]

**Screenshots**
See abive

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser : is this relevant?
 - Version: 10

**Additional context**
I am using ``robotology-superbuild`` with conda dependencies

cc @kouroshD

S-Dafarra, 2022-04-04:
The problem seems that the ``SGCoreCpp.dll`` is not found: 
[image]

kouroshD, 2022-04-04:
@S-Dafarra  please follow the documentation in https://github.com/robotology/wearables/blob/master/doc/How-to-compile-and-run-HapticGlove.md as indicated. This should solve your problem.

S-Dafarra, 2022-04-04:
Yes, I forgot to add ``D:\Windows\SenseGlove-API\Core\SGCoreCpp\lib\win\Release`` to the PATH. Fixed, thanks

closed 2022-04-04 by S-Dafarra