robotology/wearables #164: CMake error when an already configured wearables is re-configured

traversaro, 2022-09-15:
**Describe the bug**

If I run `cmake .` again in an already created built environment, the configuration fails with:
~~~
-- Found YARP: /home/gitpod/mambaforge/envs/wearablesdev/lib/cmake/YARP (found version "3.7.2~dev")
--  [x] Plugin: iwear_logger (ENABLE_iwear_logger)
CMake Error at modules/IWearFrameVisualizer/CMakeLists.txt:22 (find_package):
  By not providing "FindIWear.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "IWear", but
  CMake did not find one.

  Could not find a package configuration file provided by "IWear" with any of
  the following names:

    IWearConfig.cmake
    iwear-config.cmake

  Add the installation prefix of "IWear" to CMAKE_PREFIX_PATH or set
  "IWear_DIR" to a directory containing one of the above files.  If "IWear"
  provides a separate development package or SDK, be sure it has been
  installed.
~~~

**To Reproduce**

~~~
mamba create -n wearablesdev idyntree librobometry yarp-cxx cmake pkg-config compilers
mamba activate wearablesdev 
git clone https://github.com/robotology/wearables
cd wearables
mkdir build
cd build
cmake ..
cmake .
~~~

**Expected behavior**

The project should not fail at the second configuration.

**Screenshots**

N.a. .

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04, dependencies installed with conda

traversaro, 2022-09-15:
The problem is that the option `ENABLE_FrameVisualizer` is defined after the related directory is added https://github.com/robotology/wearables/blob/56c3db149f073ef4ff54322a11c5f2bd7364cbed/CMakeLists.txt#L103 . Furthermore, it is not clear why the FrameVisualizer calls `find_package(IWear)` if the IWear is created by the CMake project, as at the first run for sure it is not available.

lrapetti, 2022-09-22:
> The problem is that the option `ENABLE_FrameVisualizer` is defined after the related directory is added

This problem seems to apply to all the options defined in https://github.com/robotology/wearables/blob/b152236ea625f2dc930ccadca2e99ce1d4b06b73/CMakeLists.txt#L95-L110, am I right?

traversaro, 2022-09-22:
> , am I right?

Yes

closed 2023-02-06 by RiccardoGrieco