robotology/wearables #177: Compilation failure against YARP master (future YARP 3.8) November 2022

traversaro, 2022-11-17:
In YARP 3.3 (released in 2019) the following headers where deprecated (see https://www.yarp.it/latest/v3_3_0.html#autotoc_md512):

| Deprecated Header | Replacement |
|:-----------------:|:------------:|
| `yarp/dev/SerialInterfaces.h` | `yarp/dev/ISerialDevice.h` |
| `yarp/dev/PreciselyTimed.h` | `yarp/dev/IPreciselyTimed.h` |
| `yarp/dev/GenericSensorInterfaces.h` | `yarp/dev/IGenericSensor.h` |
| `yarp/dev/Wrapper.h` | `yarp/dev/IWrapper.h` and/or `yarp/dev/IMultipleWrapper.h` | 

However, the inclusion of the deprecated headers, so probably the deprecation went unnoticed. However, in YARP master branch (future YARP 3.8) the headers were removed, so it would be great to remove the use of this deprecated headers, as it has been detected that this repo is using those headers.

Cross-ref : https://github.com/robotology/robotology-superbuild/issues/1303

lrapetti, 2022-11-28:
Fixed in https://github.com/robotology/wearables/pull/179

closed 2022-11-28 by lrapetti

traversaro, 2022-11-28:
Thanks @lrapetti !