robotology/wearables #181: IWearLogger not starting on Windows

Gianlucamilani, 2023-02-06:
I am trying to run the IWearLogger device on Windows, and it is failing with:
~~~
[DEBUG] IWearRemapper : Opened correctly
[INFO] |yarp.dev.PolyDriver|iFeelRemapper| Created device <iwear_remapper>. See C++ class wearable::devices::IWearRemapper for documentation.
[INFO] Opening device iFeelLoggerDevice with parameters [("robotName" = ""), ("period" = "0.01"), ("logAllQuantities" = "true"), ("logAccelerometers" = "false"), ("logEMGSensors" = "false"), ("logForce3DSensors" = "false"), ("logForceTorque6DSensors" = "false"), ("logFreeBodyAccelerationSensors" = "false"), ("logGyroscopes" = "false"), ("logMagnetometers" = "false"), ("logOrientationSensors" = "false"), ("logPoseSensors" = "false"), ("logPositionSensors" = "false"), ("logTemperatureSensors" = "false"), ("logTorque3DSensors" = "false"), ("logVirtualLinkKinSensors" = "false"), ("logVirtualJointKinSensors" = "false"), ("logVirtualSphericalJointKinSensors" = "false"), ("logSkinSensors" = "false"), ("saveBufferManagerConfiguration" = "true"), ("experimentName" = "ifeel_data"), ("path" = "/tmp/ifeel_data/"), ("n_samples" = "100000"), ("save_periodically" = "true"), ("save_period" = "120.0"), ("data_threshold" = "300"), ("auto_save" = "true")]
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/Users/gimilani/Documents/Github/wearables/build/install/lib/yarp/IWearLogger.dll:
  The specified module could not be found.

[WARNING] |yarp.os.YarpPluginSettings| Cannot load plugin from shared library (IWearLogger)
[WARNING] |yarp.os.YarpPluginSettings| (The specified module could not be found.

The specified module could not be found.

The specified module could not be found.
)
[ERROR] |yarp.dev.PolyDriver|iFeelLoggerDevice| Could not find device <iwear_logger>
[WARNING] Cannot open device iFeelLoggerDevice
[WARNING] Cannot open device iFeelLoggerDevice
~~~

Debugging with dep walker, I have this problem:
[image]

So for some reason YARP_telemetry is missing, the packages I have installed with mamba are:
~~~
(robotologyenv) C:\Users\gimilani>mamba list
# packages in environment at C:\Users\gimilani\AppData\Local\mambaforge\envs\robotologyenv:
#
# Name                    Version                   Build  Channel
ace                       7.0.9                h63175ca_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
assimp                    5.2.4                h4dcb625_2    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.5.2                hcfcfb64_0    conda-forge
ca-certificates           2022.12.7            h5b45459_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.25.2               h1537add_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.5.2                h57928b3_0    conda-forge
cxx-compiler              1.5.2                h91493d7_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h1537add_0    conda-forge
ffmpeg                    5.1.2           gpl_h5b1d025_106    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.5.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.74.1               h12be248_1    conda-forge
glib-tools                2.74.1               h12be248_1    conda-forge
graphite2                 1.3.13                     1000    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gst-plugins-base          1.21.3               h001b923_1    conda-forge
gstreamer                 1.21.3               h6b5321d_1    conda-forge
[... 259 lines cut ...]
[INFO] |yarp.dev.Drivers| Device "rangefinder2D_nws_ros", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_rangefinder2D_nws_ros.dll library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "rangefinder2D_nws_yarp", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_rangefinder2D_nws_yarp.dll library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "remote_controlboard", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_remote_controlboard.dll library), wrapped by "controlboardwrapper2".
[INFO] |yarp.dev.Drivers| Device "remote_grabber", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_remote_grabber.dll library), wrapped by "grabber".
[INFO] |yarp.dev.Drivers| Device "remotecontrolboardremapper", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_controlboardremapper.dll library), wrapped by "controlboardwrapper2".
[INFO] |yarp.dev.Drivers| Device "rgbdSensor_nws_ros", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_rgbdSensor_nws_ros.dll library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "rgbdSensor_nws_yarp", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_rgbdSensor_nws_yarp.dll library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "rgbdToPointCloudSensor_nws_ros", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_rgbdToPointCloudSensor_nws_ros.dll library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "robotDescriptionClient", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_robotDescriptionClient.dll library).
[INFO] |yarp.dev.Drivers| Device "robotDescriptionServer", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_robotDescriptionServer.dll library).
[INFO] |yarp.dev.Drivers| Device "serial", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_serial.dll library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "serialport", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_serialport.dll library), wrapped by "serial".
[INFO] |yarp.dev.Drivers| Device "telemetryDeviceDumper", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_telemetryDeviceDumper.dll library).
[INFO] |yarp.dev.Drivers| Device "transformClient", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/yarp_transformClient.dll library).
[... cut ...]
[INFO] |yarp.dev.Drivers| Device "iwearactuators_wrapper", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IWearActuatorsWrapper.dll library).
[INFO] |yarp.dev.Drivers| Device "ixsensmvncontrol_wrapper", available on request (found in C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IXsensMVNControlWrapper.dll library).
[INFO] |yarp.dev.Drivers| Device "ifeelsuit", available on request (found in C:/Users/gimilani/Documents/Github/component_ifeel/element_wearable_sw/build/install/lib/yarp/iFeelSuit.dll library).
[INFO] |yarp.dev.Drivers|
~~~

traversaro, 2023-02-06:
Ok, I fixed one related problem in https://github.com/robotology/wearables/pull/182 . However, in this specific case the problem is not in the `conda` packages of wearables: instead, the problem is that a `wearables` is installed in the environment, but YARP actually finds the wearables devices in a local install of wearables in `C:/Users/gimilani/Documents/Github/wearables/build/install/` .

So the possible options are:
* Uninstall the wearables installed via conda, and update and re-compile the wearables installed in `C:/Users/gimilani/Documents/Github/wearables/build`.
* Remove the local wearables from   `C:/Users/gimilani/Documents/Github/wearables/`, and just wait for the new release of wearables to be released and available as a conda package.

traversaro, 2023-02-06:
Reopening until we have updated conda packages.

RiccardoGrieco, 2023-02-06:
> Reopening until we have updated conda packages.

Sorry I misclicked 😅

traversaro, 2023-02-06:
> > Reopening until we have updated conda packages.
> 
> Sorry I misclicked 😅

No problem, this was closed by the `Fix`  command in the PR body.

traversaro, 2023-02-07:
We have 1.7.1 in https://anaconda.org/robotology/wearables/files, so everything should work.

Gianlucamilani, 2023-02-10:
- Together with @RiccardoGrieco I tried to update wearables 
- The package correctly updates at version 1.7.1:

```
wcwidth                   0.2.6              pyhd8ed1ab_0    conda-forge
wearables                 1.7.1           py310h595d6f7_75    robotology
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
```

- We then tried to launch the module:

> (robotologyenv) C:\Users\gimilani>yarprobotinterface --config iFeelSuit_logger.xml

- But this error shows off:

> [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IWearLogger.dll:

- Actually, if you try to run `yarpdev --list` the file is not found and this error message appear:

> [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IWearLogger.dll:

- BUT IN MY PC THE FILE IS PRESENT inside the folder C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IWearLogger.dll:

CC @traversaro

traversaro, 2023-02-10:
Can you provide the complete mamba list of your environment? On my system with wearable 1.7.1 yarpdev --list was working correctly.

traversaro, 2023-02-10:
> > Error while opening C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IWearLogger.dll: The specified module could not be found.
>BUT IN MY PC THE FILE IS PRESENT inside the folder C:/Users/gimilani/AppData/Local/mambaforge/envs/robotologyenv/Library/lib/yarp/IWearLogger.dll

Sorry, that error is misleading and should probably be changed. The error is printed when the library can't be loaded even if the .dll file exists and is found. Tipically, it can't be loaded if some required library is missing. To debug why it can't be loaded one can use  https://github.com/lucasg/Dependencies or other similar tools.

Gianlucamilani, 2023-02-13:
here my complete mamba list @traversaro 

```
# Name                    Version                   Build  Channel
ace                       7.0.11               h63175ca_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.5.2                hcfcfb64_0    conda-forge
ca-certificates           2022.12.7            h5b45459_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.25.2               h1537add_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.5.2                h57928b3_0    conda-forge
cxx-compiler              1.5.2                h91493d7_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h1537add_0    conda-forge
ffmpeg                    5.1.2           gpl_h5b1d025_106    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.5.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.74.1               h12be248_1    conda-forge
glib-tools                2.74.1               h12be248_1    conda-forge
graphite2                 1.3.13                     1000    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gst-plugins-base          1.22.0               h001b923_0    conda-forge
gstreamer                 1.22.0               h6b5321d_0    conda-forge
harfbuzz                  6.0.0                he256f1b_0    conda-forge
hdf5                      1.12.1          nompi_h57737ce_104    conda-forge
icu                       70.1                 h0e60522_0    conda-forge
icub-firmware-shared      1.29.0              h63175ca_75    robotology
icub-main                 2.0.2           py310h33d2085_75    robotology
idyntree                  8.1.0           py310h666c5ba_0    conda-forge
idyntree-matlab-bindings  8.1.0               h63175ca_75    robotology
intel-openmp              2023.0.0         h57928b3_25922    conda-forge
ipopt                     3.14.10              ha9547d1_0    conda-forge
ipython                   8.9.0              pyh08f2357_0    conda-forge
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jasper                    2.0.33               hc2e4405_1    conda-forge
jedi                      0.18.2             pyhd8ed1ab_0    conda-forge
jpeg                      9e                   h8ffe710_2    conda-forge
krb5                      1.20.1               heb0366b_0    conda-forge
lcms2                     2.14                 ha5c8aab_1    conda-forge
lerc                      4.0.0                h63175ca_0    conda-forge
libaec                    1.0.6                h63175ca_1    conda-forge
libblas                   3.9.0              16_win64_mkl    conda-forge
libcblas                  3.9.0              16_win64_mkl    conda-forge
libclang                  15.0.7          default_h77d9078_0    conda-forge
libclang13                15.0.7          default_h77d9078_0    conda-forge
libcurl                   7.87.0               h68f0423_0    conda-forge
libdeflate                1.17                 hcfcfb64_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
[... 85 lines cut ...]
vs2015_runtime            14.34.31931         h4c5c07a_10    conda-forge
vs2019_win-64             19.29.30139         hb9aee9d_10    conda-forge
vswhere                   3.1.1                h57928b3_0    conda-forge
wcwidth                   0.2.6              pyhd8ed1ab_0    conda-forge
wearables                 1.7.1           py310h595d6f7_75    robotology
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
[... cut ...]
yarp-python               3.7.2           py310ha1c043d_8    conda-forge
ycm-cmake-modules         0.15.1               h63175ca_1    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.13               hcfcfb64_4    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge
```

> To debug why it can't be loaded one can use https://github.com/lucasg/Dependencies or other similar tools.

I can try by myself?

Gianlucamilani, 2023-02-13:
tomorrow is perfect for me @traversaro

traversaro, 2023-02-13:
> tomorrow is perfect for me @traversaro thanks

In the meanwhile, can you try to "create a new environment from scratch with just the packages you use?", as mentioned in the previous message? If that works, we do not need need to use Dependencies at all!

Gianlucamilani, 2023-02-14:
Created a new environment thanks to the help of @RiccardoGrieco with:

> mamba install -c conda-forge -c robotology yarp wearables=1.7.1 idyntree icub-main yarp-matlab-bindings  idyntree-matlab-bindings icub-firmware-shared

here the mamba list output

```
#
# Name                    Version                   Build  Channel
ace                       7.0.11               h63175ca_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.5.2                hcfcfb64_0    conda-forge
ca-certificates           2022.12.7            h5b45459_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.25.2               h1537add_0    conda-forge
compilers                 1.5.2                h57928b3_0    conda-forge
cxx-compiler              1.5.2                h91493d7_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.5.0                h1537add_0    conda-forge
ffmpeg                    5.1.2           gpl_h5b1d025_106    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.5.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.74.1               h12be248_1    conda-forge
glib-tools                2.74.1               h12be248_1    conda-forge
graphite2                 1.3.13                     1000    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gst-plugins-base          1.22.0               h001b923_0    conda-forge
gstreamer                 1.22.0               h6b5321d_0    conda-forge
harfbuzz                  6.0.0                he256f1b_0    conda-forge
hdf5                      1.14.0          nompi_h5af7005_102    conda-forge
icu                       70.1                 h0e60522_0    conda-forge
icub-firmware-shared      1.29.0              h63175ca_76    robotology
icub-main                 2.0.2           py310h70c1692_76    robotology
idyntree                  8.1.0           py310h666c5ba_0    conda-forge
idyntree-matlab-bindings  8.1.0               h63175ca_76    robotology
intel-openmp              2023.0.0         h57928b3_25922    conda-forge
ipopt                     3.14.10              ha9547d1_0    conda-forge
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jasper                    2.0.33               hc2e4405_1    conda-forge
jpeg                      9e                   hcfcfb64_3    conda-forge
krb5                      1.20.1               heb0366b_0    conda-forge
lerc                      4.0.0                h63175ca_0    conda-forge
libblas                   3.9.0              16_win64_mkl    conda-forge
libcblas                  3.9.0              16_win64_mkl    conda-forge
libclang                  15.0.7          default_h77d9078_1    conda-forge
libclang13                15.0.7          default_h77d9078_1    conda-forge
libcurl                   7.87.0               h68f0423_0    conda-forge
libdeflate                1.17                 hcfcfb64_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libglib                   2.74.1               he8f3873_1    conda-forge
libhwloc                  2.8.0                h039e092_1    conda-forge
libiconv                  1.17                 h8ffe710_0    conda-forge
liblapack                 3.9.0              16_win64_mkl    conda-forge
liblapacke                3.9.0              16_win64_mkl    conda-forge
[... 54 lines cut ...]
tzdata                    2022g                h191b570_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
vc                        14.3                hb6edc58_10    conda-forge
visit_struct              1.0.0.1              h0e60522_1    conda-forge
vs2015_runtime            14.34.31931         h4c5c07a_10    conda-forge
vs2019_win-64             19.29.30139         hb9aee9d_10    conda-forge
vswhere                   3.1.1                h57928b3_0    conda-forge
wearables                 1.7.1           py310h595d6f7_76    robotology
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
[... cut ...]
yarp-cxx                  3.7.2                h8a0a163_9    conda-forge
yarp-matlab-bindings      3.7.3               h63175ca_76    robotology
yarp-python               3.7.2           py310ha1c043d_9    conda-forge
ycm-cmake-modules         0.15.1               h63175ca_1    conda-forge
zlib                      1.2.13               hcfcfb64_4    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge

```

I then recompiled component-ifeel as per https://github.com/ami-iit/component_ifeel/tree/master/element_wearable_sw

@traversaro

Gianlucamilani, 2023-02-14:
Now I try to launch the application @traversaro

Gianlucamilani, 2023-02-14:
Now it seems to work @traversaro @RiccardoGrieco

traversaro, 2023-02-14:
Great! So we can close the issue?

Gianlucamilani, 2023-02-14:
Yes thanks @traversaro

closed 2023-02-14 by traversaro