robotology/wearables #184: CI Warnings

traversaro, 2023-02-06:
~~~
Warning: The `save-state` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Warning: The `save-state` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Cache Size: ~8 MB (8032484 B)
/usr/bin/tar -xz -f /Users/runner/work/_temp/66e2fe7d-34ec-43fb-ae8d-46e62aa75011/cache.tgz -C /Users/runner/work/wearables/wearables/install/deps
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Cache restored from key: source-deps-macOS-/Users/runner/work/_temp-vcpkg-robotology-v0.7.0-ycm-v0.14.0-yarp-v3.5.0-iDynTree-v5.1.0-matioCpp-v0.2.0-robometry-v1.2.0
~~~

RiccardoGrieco, 2023-03-16:
Is this still happening? I don't see it in the latest CI checks https://github.com/robotology/wearables/actions/runs/4432905929.

traversaro, 2023-03-16:
Seems to be solved indeed.

closed 2023-03-16 by traversaro