robotology/wearables #188: Quaternion/Position order is not coherent between logger and thrift

lrapetti, 2023-04-04:
**Describe the bug**
With @Zweisteine96 we noticed difference in the linear/angular serialization used used for iFeel data:
- in the thrift message we use angular/linear for the pose (see [here](https://github.com/robotology/wearables/blob/59ea45c9c80dec9317790c5a4b8712a31a7d9ee1/msgs/thrift/WearableData.thrift#L55-L56) and [here](https://github.com/robotology/wearables/blob/59ea45c9c80dec9317790c5a4b8712a31a7d9ee1/msgs/thrift/WearableData.thrift#L60-L61)), while for the velocity in the thift we use linear/angular (see [here](https://github.com/robotology/wearables/blob/59ea45c9c80dec9317790c5a4b8712a31a7d9ee1/msgs/thrift/WearableData.thrift#L62-L63))
- In the sensors interface implementation the order seems to be again orientation/position (see [here](https://github.com/robotology/wearables/blob/efb34b1baaf98006ef2e9add0cdfe98ec6f30e86/interfaces/IWear/include/Wearable/IWear/Sensors/IPoseSensor.h#L32) and [here](https://github.com/robotology/wearables/blob/efb34b1baaf98006ef2e9add0cdfe98ec6f30e86/interfaces/IWear/include/Wearable/IWear/Sensors/IPoseSensor.h#L48-L65))
- In the logger, data are saved as position/orientation (see [here](https://github.com/robotology/wearables/blob/59ea45c9c80dec9317790c5a4b8712a31a7d9ee1/wrappers/IWearLogger/src/IWearLogger.cpp#L522-L530) and [here](https://github.com/robotology/wearables/blob/59ea45c9c80dec9317790c5a4b8712a31a7d9ee1/wrappers/IWearLogger/src/IWearLogger.cpp#L663-L664))

The difference between logger and thrift has created some confusion.

This opens ups to some solutions:
- Document property the way data are logged and transmitted
- Change the order in the thrift (this will break backcompatibility of the dataset, even if it would be clear the order used)
- Change the order in the logger (same as previous)
- ... others?

Any suggestion/opinion @RiccardoGrieco @traversaro @diegoferigo ?

traversaro, 2023-04-04:
If we want to change for harmonization, I would stick to linear/angular serialization as much as possible. Recently I also investigated integrating iDynTree and pinocchio in https://github.com/ami-iit/idynfor, and also in pinocchio the serialization used is linear/angular, see https://github.com/ami-iit/idynfor/blob/97ff77e6f148cabe7d95f51ee1c67958e9c9f918/doc/theory_background.md#rigid-body-velocity .

lrapetti, 2023-04-04:
> If we want to change for harmonization, I would stick to linear/angular serialization as much as possible

Yess, in general I agree with this. It would actually be a very quick change in the code.

The main issue, however, is that it would break all the datasets that we logged in the past using the `yarpdatdumper` which is the default way to store datasets. I don't know if there is a way to have a smooth transition

traversaro, 2023-04-04:
> The main issue, however, is that it would break all the datasets that we logged in the past using the `yarpdatdumper` which is the default way to store datasets. I don't know if there is a way to have a smooth transition.

If we have them saved all somewhere, we could write a simple tool to convert the datasets between the old and the new version. But I do not know if doing that is worth the time/effort necessary to do that.