robotology/wearables #190: Problem in opening and publishing data on a wearable port using python bindings

lrapetti, 2023-06-27:
With @Zweisteine96 we tried to use the python bindings (that were added with https://github.com/robotology/wearables/pull/162) in order to open a wearable data port and publish some data.

This is the script we are using

```py
import yarp
import wearables.bindings as wearables
import time

# create the yarp network 
network = yarp.Network()
network.init()

port = wearables.msg.BufferedPortWearableData()
port.open("/foo2")

time.sleep(1)

data = port.prepare()
data.producerName = "myWearableDevice"

time.sleep(1)

# here we create a PoseSensorData message, and try to update the value of the position x
myPoseSensorData = wearables.msg.PoseSensorData()
myPoseSensorData.position.x = 1.0
print(myPoseSensorData.position.x) #this returns the expected value

# here we want to add myPoseSensorData to the data.poseSensors dictionary
poseSensors = data.poseSensors
poseSensors.update({"firstTracker":myPoseSensorData})

# If we try to print the value from the dictionary here it returns an error (see below)
# print(data.poseSensors["firstTracker"].position.x)

while True:
  print(data)
  port.write()

  time.sleep(1)
```

But what happen is that:
- if we try to access the data from the dictionary (the line that is commented above) we get the following error:
  ```
  Traceback (most recent call last):
  File "wearable_data_generator.py", line 33, in <module>
    print(data.poseSensors["firstTracker"].position.x)
  ```

- If instead we leave that line commented, when printing `data` it seems that the data is not added (this is what it is printed in the while loop by `print(data)`):
  ```sh
  myWearableDevice () () () () () () () () () () () () () () () ()
  ```
  (We can observe that the `data.producerName` is updated, but the rest is empty)

lrapetti, 2023-06-27:
Maybe @traversaro or @GiulioRomualdi that have more experience on bindings generation and usage might help?

traversaro, 2023-06-27:
Indeed, this is quite counterintuitive, but I guess it is how pybind11 works, see https://pybind11.readthedocs.io/en/stable/advanced/cast/stl.html : 

> In this case, properties can be read and written in their entirety. However, an append operation involving such a list type has no effect:

So, we canno append to the `poseSensors` attribute, but only substitute it. Accounting for this, and for the element type of the `poseSensors` map is the `PoseSensors`, not `PoseSensorsData` (see https://github.com/robotology/wearables/blob/v1.7.1/msgs/thrift/WearableData.thrift#L183), a working variant of your script is:
~~~
import yarp
import wearables.bindings as wearables
import time

# create the yarp network 
network = yarp.Network()
network.init()

port = wearables.msg.BufferedPortWearableData()
port.open("/foo2")

time.sleep(1)

data = port.prepare()
data.producerName = "myWearableDevice"

time.sleep(1)

# here we create a PoseSensorData message, and try to update the value of the position x
myPoseSensorData = wearables.msg.PoseSensorData()
myPoseSensorData.position.x = 1.0
print(myPoseSensorData.position.x) #this returns the expected value
myPoseSensor = wearables.msg.PoseSensor()
myPoseSensor.data = myPoseSensorData

# here we want to add myPoseSensor to the data.poseSensors dictionary
originalPoseSensors = data.poseSensors
originalPoseSensors.update({"firstTracker": myPoseSensor})
data.poseSensors = originalPoseSensors

print(data)

# If we try to print the value from the dictionary here it returns an error (see below)
print(data.poseSensors["firstTracker"].data.position.x)

while True:
  print(data)
  port.write()

  time.sleep(1)
~~~

Tested with:
~~~
~~~

lrapetti, 2023-06-28:
Thanks a lot @traversaro! The example you provided is working for me and I see the following print of `data`:
```sh
myWearableDevice () () () () () () () () ((firstTracker ("" 0 0.0 0.0 0.0 0.0 1.0 0.0 0.0))) () () () () () () ()
```

Still I am not able to read the port from `yarp`
```sh
$ yarp read ... /foo2
\[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.0.2.116:10003/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /foo2 to /tmp/port/1 using tcp
```
this was happening also with my previous script, it was not publishing even the empty message.

traversaro, 2023-06-29:
Working version:
~~~
import yarp
import wearables.bindings as wearables
import time

# create the yarp network 
network = yarp.Network()
network.init()

port = wearables.msg.BufferedPortWearableData()
port.open("/foo2")

time.sleep(1)

data = port.prepare()
data.producerName = "myWearableDevice"

time.sleep(1)

# here we create a PoseSensorData message, and try to update the value of the position x
myPoseSensorData = wearables.msg.PoseSensorData()
myPoseSensorData.position.x = 1.0
print(myPoseSensorData.position.x) #this returns the expected value
myPoseSensor = wearables.msg.PoseSensor()
myPoseSensor.data = myPoseSensorData

# here we want to add myPoseSensor to the data.poseSensors dictionary
originalPoseSensors = data.poseSensors
originalPoseSensors.update({"firstTracker": myPoseSensor})
data.poseSensors = originalPoseSensors

print(data)

# If we try to print the value from the dictionary here it returns an error (see below)
print(data.poseSensors["firstTracker"].data.position.x)

dataToSend = data

while True:
  newData = port.prepare()
  newData = dataToSend
  print(dataToSend)
  print(newData)
  port.write()

  time.sleep(1)

~~~

lrapetti, 2023-06-30:
Thanks @traversaro, I am now able to read data from the port, what was missingwas to capp `port.prepare()` in the loop.

Here, for reference, is the final test script I was using for testing:

```py
import yarp
import wearables.bindings as wearables
import time

# create the yarp network 
network = yarp.Network()
network.init()

# create a buffered wearable data port
port = wearables.msg.BufferedPortWearableData()
port.open("/foo")

time.sleep(0.2)

# prepare data
dataToSend = port.prepare()
dataToSend.producerName = "myWearableDevice"

time.sleep(0.2)

# prepare a pose sensor data
myPoseSensor = wearables.msg.PoseSensor()
myPoseSensor.data = wearables.msg.PoseSensorData()
myPoseSensor.info.name = "myWearableDevice::Pose::FirstTracker"
myPoseSensor.info.status = wearables.msg.OK

# add the pose sensor data to the data.poseSensors dictionary
originalPoseSensors = dataToSend.poseSensors
dataToSend.poseSensors.update({"FristTracker": myPoseSensor})
dataToSend.poseSensors = originalPoseSensors

count = 0

while True:
  # update data 
  count = count + 1
  dataToSend.poseSensors["FristTracker"].data.position.x = count

  data = port.prepare()
  data = dataToSend

  print(data)

  port.write()

  time.sleep(0.05)
```

cc @Zweisteine96 I think this will be useful for your use case.

closed 2023-06-30 by lrapetti