robotology/wearables #191: ``frameTransformClient`` is going to be removed in the next release of YARP

S-Dafarra, 2023-08-02:
The ``frameTransformClient`` device is going to be removed in the next version of YARP and thus we need to update the code using it.
See https://github.com/robotology/yarp/pull/2999

Related issues: 
- https://github.com/ami-iit/yarp-device-openxrheadset/issues/40
- https://github.com/robotology/walking-teleoperation/issues/131

cc @lrapetti

lrapetti, 2023-08-24:
Thanks, @S-Dafarra for raising it. Actually it is the `transformClient` that will be removed and the `frameTransformClient` should be the replacement, did I understand it correctly (https://github.com/robotology/yarp/pull/2999#issuecomment-1662155403)?

S-Dafarra, 2023-08-24:
Yes you are right!

lrapetti, 2023-08-24:
> Yes you are right!

Thanks! I have changed the title of the issue accordingly

S-Dafarra, 2023-10-04:
I think it should be sufficient to edit https://github.com/robotology/wearables/blob/3fe3182e2ecec116549c0556baf312a9edc07ff6/devices/IFrameTransformToIWear/conf/iframetransform_to_iwear.xml#L5-L9 right?

S-Dafarra, 2023-11-03:
These are the lines to use:
```xml
<device type="frameTransformClient" name="TransformClient">
    <param name="period">0.01</param>
    <param name="filexml_option">ftc_yarp_only.xml</param>
    <param name="ft_client_prefix">/IFrameTransformToIWear/tf</param>
    <param name="local_rpc">/IFrameTransformToIWear/tf/local_rpc</param>
</device>
```

We also need to edit the HDE files in https://github.com/robotology/human-dynamics-estimation/tree/77c7a024d875956ab05205171109ed1a82dc46e3/conf/xml

S-Dafarra, 2023-11-03:
With @davidegorbani we tried today but we had issues. In particular, we had this error continuosly:
```
[ERROR] |yarp.device.FrameTransformClient| getTransform(): Frames  root_link_desired  and  openxr_head  are not connected
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[ERROR] |yarp.device.FrameTransformClient| getTransform(): Frames  root_link_desired  and  openxr_head  are not connected
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_right_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[ERROR] |yarp.device.FrameTransformClient| getTransform(): Frames  root_link_desired  and  openxr_head  are not connected
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[ERROR] |yarp.device.FrameTransformClient| getTransform(): Frames  root_link_desired  and  openxr_head  are not connected
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::openxr_head
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_left_elbow_pose
[WARNING] HumanStateProvider : Failed to read link position from virtual link sensor  TransformServer::pose::vive_tracker_right_elbow_pose
[WARNING] HumanStateProvider : Failed to read link orientation from virtual link sensor  TransformServer::pose::vive_tracker_right_elbow_pose
[... cut ...]
We have the transform server running on a Windows machine, while the client on an Ubuntu machine on the same network.
I have tried to edit the timeout (specified in https://github.com/robotology/yarp/blob/65a3a508c0b5147bdf5106ac173370382486004e/src/libYARP_dev/src/yarp/dev/FrameTransformContainer.h#L99 and not set from conf file) from 0.2s to 1.0s on both machines, but nothing changed. I wonder if this problem has been fixed in recent version of ``yarp``. Currently we are using 3.7.2,  although I don't see anything particular in the changelog

S-Dafarra, 2023-11-07:
> We have the transform server running on a Windows machine, while the client on an Ubuntu machine on the same network.
> I have tried to edit the timeout (specified in [...]

This appeared to be a much deeper issue: https://github.com/robotology/yarp/issues/3038

S-Dafarra, 2023-11-09:
After applying the patch to ``yarp`` mentioned in https://github.com/robotology/yarp/pull/3041, the errors disappeared and the IK seems to be working fine. I will proceed in opening PRs to edit the configuration files

S-Dafarra, 2023-11-09:
Modifications done in:
- https://github.com/robotology/wearables/pull/202
- https://github.com/robotology/human-dynamics-estimation/pull/377

PR #202 (2023-11-14): Update iframetransform_to_iwear.xml to use new TF device
Fixes #191
closed 2023-11-14 by lrapetti