robotology/wearables #212: Merge wearables in human-dynamics-estimation?

traversaro, 2024-09-19:
Hello everyone, given that:
* Both `wearables` and `human-dynamics-estimation` do not have a well define maintainer, with people (mainly me, @S-Dafarra and @dariosortino) handling PRs and/or tags and released when needed.
* We would like to have `human-dynamics-estimation` conda packages with python bindings enabled on `linux-aarch64`, to test the recent work of team X on Nvidia SoC systems like Jetson, and the easiest way to get this done (see https://github.com/robotology/robotology-superbuild/issues/738) is to bring the related packages in conda-forge. 

I thought that to simplify mantainance of the repo, we could merge `wearables` in `human-dynamics-estimation` so that:
* They can be packaged and mantained as a single package on conda-forge
* If we need to do a change and released them, this can be done with a single PR that modify both, and if a release/tag is necessary a single one can be done.

Anyone has any specific opinion on this? If people agree, I will be glad to do the change myself. Note that if there is someone that only need wearables, I plan to add specific CMake options so that you could still build only the existing wearables code out of the human-dynamics-estimation repo.

To give you an idea of the current state of dependencies, this is the current tree of dependencies in robotology-superbuild (generated via `dot -Tpdf robotology-superbuild.dot -o view.pdf` in the build of the superbuild):

[image]

dariosortino, 2024-09-19:
I never worked directly on `wearables` but I think that for iFeel applications is not a problem. Let us know if there is something to update in whatever is used by `human-dynamics-estimation` and `wearables`

traversaro, 2024-09-24:
PR ready for the migration:
* https://github.com/robotology/human-dynamics-estimation/pull/395
* https://github.com/robotology/robotology-superbuild/pull/1720

traversaro, 2024-09-24:
The plan is:
* Get the green light of @dariosortino and @S-Dafarra (see https://github.com/robotology/human-dynamics-estimation/pull/395#issuecomment-2370637902)
* Merge https://github.com/robotology/human-dynamics-estimation/pull/395 and https://github.com/robotology/robotology-superbuild/pull/1720
* Tag human-dynamics-estimation 4.0.0 (major bump to easily detect before and after the merge)
* Package human-dynamics-estimation 4.0.0 in conda-forge (currently blocked by https://github.com/conda-forge/staged-recipes/pull/27675)

Once this is done, we should have everything to run the online inference/ML part of Team X demos in conda-forge, including `linux-aarch64` packages. fyi @GiulioRomualdi @giotherobot @carloscp3009

traversaro, 2024-09-29:
> 
>     * [ ]  Package human-dynamics-estimation 4.0.0 in conda-forge (currently blocked by [Add recipe for icub-firmware-shared packages conda-forge/staged-recipes#27675](https://github.com/conda-forge/staged-recipes/pull/27675) and [Add recipe for icub-main packages conda-forge/staged-recipes#27692](https://github.com/conda-forge/staged-recipes/pull/27692))
> 
> 

In progress in https://github.com/conda-forge/staged-recipes/pull/27723 .

traversaro, 2024-10-01:
> > ```
> > * [ ]  Package human-dynamics-estimation 4.0.0 in conda-forge (currently blocked by [Add recipe for icub-firmware-shared packages conda-forge/staged-recipes#27675](https://github.com/conda-forge/staged-recipes/pull/27675) and [Add recipe for icub-main packages conda-forge/staged-recipes#27692](https://github.com/conda-forge/staged-recipes/pull/27692))
> > ```
> 
> In progress in [conda-forge/staged-recipes#27723](https://github.com/conda-forge/staged-recipes/pull/27723) .

PR ready, just waiting for review now.

traversaro, 2024-10-01:
PR merged, the last step is to swich the conda generation pipeline for `robotology` channel to use the new packages, see https://github.com/robotology/robotology-superbuild/pull/1727 .

traversaro, 2024-10-04:
Great, with just ~5 days of delay we have `robotology` packages built against the latest hde 4.0.0 on conda-forge: https://github.com/robotology/robotology-superbuild/actions/runs/11170548014 . At this point I think I can do a nice announcement and close this issue.

traversaro, 2024-10-04:
Announcement done: https://github.com/orgs/robotology/discussions/717 .

closed 2024-10-04 by traversaro